BlueAndi/Pixelix #6: Support of OpenPixelControl (OPC) protocol.

BlueAndi, 2020-01-15:
http://openpixelcontrol.org/

Nikey, 2022-11-07:
maybe better support DDP protocol, as it is used far more often in software (e.g. hyperion, ledfx...)
http://www.3waylabs.com/ddp/

Nikey, 2022-11-08:
E1.31 is limited to around 510 LED's with 3 universes (if we want to reach 40 FPS), see here:
https://github.com/Aircoookie/WLED/wiki/E1.31-DMX

DDP is the far more capable one (around 30% more efficient), see the comparison here:
http://www.3waylabs.com/ddp/#Efficiency

When I have the choice, I would always pick DDP over E1.31 for the more fluent framerates.
When I use E1.31 on my 507 LED's-Strip, the last ~30 LED's don't refresh, so the real limit should be around 470 LED's
So If someone use 2 Matrix à 8x32 we would end up on this hard limit already (512 LED's).

Just my thoughts :)

BlueAndi, 2022-11-08:
I did not analyse it yet, so interesting links. Will have a look in it. Thanks.

BlueAndi, 2023-02-01:
https://github.com/smeighan/xLights/issues/3704

BlueAndi, 2023-02-01:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/Development/doc/PLUGINS.md#ddpplugin

closed 2023-02-01 by BlueAndi