BlueAndi/Pixelix #8: Replace SPIFFS by LittleFS

0ooYoo0, 2020-06-06:


BlueAndi, 2020-06-07:
https://community.platformio.org/t/how-to-define-littlefs-partition-build-image-and-flash-it-on-esp32/11333
https://github.com/ARMmbed/littlefs
https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS

BlueAndi, 2020-06-17:
https://github.com/espressif/arduino-esp32/issues/3765#issuecomment-643975602
https://github.com/espressif/arduino-esp32/issues/3765#issuecomment-643996450
https://github.com/espressif/arduino-esp32/issues/3765#issuecomment-644011312

BlueAndi, 2020-06-17:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/792

BlueAndi, 2020-07-30:
https://github.com/espressif/esp-idf/pull/5469

BlueAndi, 2020-09-24:
https://github.com/lorol/LITTLEFS/tree/master/examples/LITTLEFS_PlatformIO

BlueAndi, 2021-03-02:
LITTLEFS will be in ESP Arduino v2.0.0 available: https://github.com/espressif/arduino-esp32/pull/4096#issuecomment-720702901

BlueAndi, 2022-04-29:
Open:
* File browser folder support
* LittleFS needs more space, check whether the meta data per file can be configured

closed 2022-11-03 by BlueAndi