BlueAndi/Pixelix #28: DHT11 (Temp/Humidity( Plugin

flcu, 2021-03-28:
Hi
Thank you for this great software! I've enjoyed building a pixelix clock and working with the base. I've created a DHT11 plugin that displays temperature and humidity as read by a dht11 sensor (connected to IO5). I'm happy to have it reviewed and maybe included.
I've also implemented a bitcoin/usd exchange rate plugin - happy to submit that as well. TLS/SSL support in the AsyncHTTP Client library is really needed for more/better choice of data sources :-)
Let me know how you want the submissions.
Have a nice day

BlueAndi, 2021-03-28:
Hi, thanks for the praise. :-)
You can create a pull-request in the PluginDevelopment branch and we will review it (according to our coding style too :-) ).
In our PCB currently IO5 is not used, therefore we will add the plugin configurable via preprocessor. Have to think about how we handle this in the Board.h too.
Waiting for some time now for TLS/SSL support and hopefully me-no-dev will have time after esp32-arduino v2.0.0 release. I try to avoid to implement it by my own, like the AsyncHttp client. We will see. On other hand e.g. if node-red is running locally, you can request data from https sources and push it to the display.
Have a nice week!

flcu, 2021-03-31:
Something to discuss: It's becoming apparent that the DHT11 is very inaccurate. However it is cheap and well sourceable. However I've now ordered a replacement (SHT31-D). Now the question: Should I then setup a second plugin for the new sensor? Or should I rather create a TempHumid plugin and make the sensors definable at compile time? One thing to note - DHT11 is a kind of onewire protocol, the new one is i2c. 
What's your opinion?

BlueAndi, 2021-03-31:
Some months ago we thought about having a universal sensor platform separate (separate HW/SW), which push the information to the display via REST API. This idea is freezed at the moment, because lack of time.

For now I would prefer having one TempHumidPlugin, where the sensor can be choosen. In the first step at compile time, e.g. via platformio.ini and maybe later via web interface at runtime. All depends on the available flash size and right now we are 85% full, but this will increase with SSL/TLS stuff for sure.

[image]

flcu, 2021-04-13:
I've now made the basic import of the source code. How should I handle the Compile-Time switches?
Two switches (one for the module itself and one for the sensor) or one just for the sensor and compile the module without sensor if none is chosen?

Regarding the flash size: 
I've read that it's possible to setup custom partition tables to increase the available space for the program (as we are running with 4MB, 2MB Flash/2 MB Data could be an option to increase the firmware size?)

flcu, 2021-04-13:
I just realized you proposed to create a class for each sensor. 
Where should I put the TempHumidDrv class definition? Into the Plugin/Plugins folder? Or some other place?

BlueAndi, 2021-04-14:
> I've now made the basic import of the source code. How should I handle the Compile-Time switches?
Two switches (one for the module itself and one for the sensor) or one just for the sensor and compile the module without sensor if none is chosen?

Does your plugin already supports more than just the DHT-11 sensor? If the DHT-11 sensor is the only one, you won't need the compile time switch. You may come back with, after the plugin is extended. This would make it easier for the review and integration too.

> Regarding the flash size:
I've read that it's possible to setup custom partition tables to increase the available space for the program (as we are running with 4MB, 2MB Flash/2 MB Data could be an option to increase the firmware size?)

Yes, custom partitions are possible. Don't forget that because of the over-the-air update concept, you need storage size twice. And right now in the filesystem we have 35% unused from 1378241 byte. My guess is, that with integration of SSL/TLS it will be full, but we will see.

> I just realized you proposed to create a class for each sensor.
Where should I put the TempHumidDrv class definition? Into the Plugin/Plugins folder? Or some other place?

The classes in the 3rd party package are just placeholders for the 3rd party libs you will require via platformio.ini

flcu, 2021-04-15:
Okay, I will submit it like this. I do have the new sensor, but I need to get it connected and everything first.
One question: How do I easily access the "duration" available for display from inside the plugin? I'd like to switch the temperature/humidity display after 1/2 duration.

0ooYoo0, 2021-04-15:
Please have a look at the DateTimePlugin (updateDateTime()) this plugin also switches its content after 1/2 duration.

closed 2021-05-02 by BlueAndi