BlueAndi/Pixelix #29: Automatic DST switching

flcu, 2021-03-28:
In order to be able to leave the clock running 24x7x365 without any maintenance, I suggest to implement automatic DST detection and switching.

Several options exist:
- Using an external service
- Reimplement: Putting a timezone JSON on the clock and allow selection of current TZ
- Using a library to implement the switch
- ?

- Check once an hour if DST has changed

BlueAndi, 2021-03-28:
Will have a look what's the best here. I know that e.g. https://github.com/nayarsystems/posix_tz_db provides a json file with all necessary data, which could be used via configTzTime() in the ClockDrv module.

BlueAndi, 2021-03-29:
Implemented the possibility to use the POSIX timezone string. Currently there is no drop-down where to choose the timezone, but may come in the future. Currently the settings are generic, without exceptions in the corresponding HTML code. Have to think about how to integrate it there.

BlueAndi, 2021-04-21:
Unfortunately the usage of timezone strings causes now in the SunrisePlugin a problem with the conversion from gmt to localtime.
The problem is in the newlib, where localtime() is located. Issue opened: https://github.com/espressif/newlib-esp32/issues/8

BlueAndi, 2022-06-20:
Fixed in https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1-RC1