BlueAndi/Pixelix #39: MicroPixelix - porting to TTGO-T-Display with output to LCD

flcu, 2021-05-01:
As the space on my desktop is limited and I did not feel like building another Pixelix (mostly because of the case-printing-time), I've implemented pixelix on a [ttgo t-display](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3) to extend the software easier (my first and so-far only built version was given away as a gift).

[image]

I just want to use this issue to discuss this implementation and keep notes.

A few things so far:
- board-specific Board.h/Pin-Definitions needed (maybe also helpful for other things, like the next sht3x sensor I'm going to implement next)
- How to use new LedMatrix instead of the stripe-based one?
- around 2% additional Flash needed (not an issue for TTGO as it is a 16MB ESP32 Flash Version)

Do you think this is even useful/worthwhile to "ever" include in the main-line?

BlueAndi, 2021-05-02:
Original Pixelix was planned to run on a LED matrix, but its a interesting idea to support other displays as well.
For this the current architecture may need some updates :-) You already mentioned the LedMatrix class.

Do you have the sourcecode for TTGO in a separate branch, where we can have a look and think about how we could support this in a more general way?

I think of course its useful and worthwhile if we can integrate and support it. Good work!

flcu, 2021-05-02:
Thanks.

Yes, I'm keeping the source-code in my [ttgo branch](https://github.com/flcu/esp-rgb-led-matrix/tree/ttgo). it's working quite well, buttons are working and the colors are correct now. I needed to redefine a few pins, but apart from that, it was quite straightforward. Brightness control is not yet implemented (should be possible using PWM).

I'm looking forward for any suggestions and I'm happy to implement/change/refactor/maintain the code to get it up to standard ;-)

BlueAndi, 2021-05-05:
I started to update the PluginDevelopment branch with the goal to have different HALs, which can be assigned via platformio.ini to a environment. If its ok for you, I will try to take over the necessary parts from your ttgo branch. I need anyway a 2nd HAL to check whether its my idea working or not.

The specific HAL contains the Board and the "LedMatrix". The LedMatrix will be transformed to a Display class, based on display interface class.

[image]

The diagram shows how it will look like.

Regarding brightness control, we can use the Color class intensity attribute in the first step.

flcu, 2021-05-05:
Looks like a good design, feel free to use my code anyway you wish, I will test it whenever you need me to.
Can I later adapt this design also for the temperature sensor (I'm now ready to implement the SHT3x sensor (which is using i2c and not a single pin)

BlueAndi, 2021-05-05:
I will notify via this issue, if its ready for testing. Regarding the sensor, to be honest I am not sure right now. But we will find a solution. :-)

flcu, 2021-05-05:
you can also see it in the ttgo branch (sorry for that...), I've implemented the SHT3x sensor using #IFDEFs for now

flcu, 2021-05-05:
#44 opened to track the sensor

BlueAndi, 2021-05-06:
@flcu The PluginDevelopment branch contains now all changes to support the TTGO T-Display board. Brightness control is done via the Color intensity attribute.

You will find in /lib/HalTtgoTDisplay the specific Board/Display classes. I removed the unnecessary pins/stuff from the Board.h

flcu, 2021-05-08:
@BlueAndi it worked out of the box 👍 
I did not yet test the brightness control, will do so asap.

BlueAndi, 2021-05-08:
After having a look again over brightness control, I found a bug, which is fixed now. Please resynchronize.

BlueAndi, 2021-05-25:
I think we can close it, the TTGO T-Display ESP32 board support is in.

closed 2021-05-25 by BlueAndi