BlueAndi/Pixelix #43: Change od Text Clour for just text plugin

franki29, 2021-05-05:
Hi, first of all a great software.
Maybe I too stupid to find, but is there any way to change the color of the text when I am using the just text plugin?

Thanks for helping

Frank

BlueAndi, 2021-05-05:
Hi Frank,
you can change the text color by using the text prefix
```\#RRGGBBYourText```
The single backslash forces the software to try to interpret the following characters. Note, for changing the color via sourcecode you must have two backslashes in the string. One additional for escaping.
The '#' is the begin of a color,
RR = Red value in hex
GG = Green value in hex
BB = Blue value in hex

Example:
```\#ff0000Hi!"```
[image]

The description is currently only in the sourcecode:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/4b2b2ac22e40b4ee8b41ee60c1725e4f5634e41a/lib/Gfx/TextWidget.h#L60-L73

franki29, 2021-05-06:
Hi, thanks for your information, but it does not work at my side.
I am using node red to send text to the display.
As soon I add \\#ff0000 to the message, it will display nothing.

Same for:
curl -X POST http://192.168.2.95/rest/api/v1/display/uid/19652/text?show=\\#FF0000Testgadsghdasghdsaghdsaghadsghdsaghsadgh
Return is:
{
  "data": {},
  "status": 0
Sure I am doing something wrong, but what?

0ooYoo0, 2021-05-06:
Please be aware that you have to replace unsafe ASCCII characters by the respective percent encoding within your url
Try to use %5C instead of \ and %23 instead of #.

See also [ASCII Encoding Reference](https://www.w3schools.com/tags/ref_urlencode.ASP)

franki29, 2021-05-06:
Ahh, yes, that is working.
Thank you very much

Br. Frank

closed 2021-05-06 by franki29

wendtc, 2021-05-06:
> 
> 
> Hi, first of all a great software.
> Maybe I too stupid to find, but is there any way to change the color of the text when I am using the just text plugin?
> 
> Thanks for helping
> 
> Frank

Hallo Frank,

Kannst Du Deine Node Red Nodes teilen - ich bin auch gerade dabei und hab viele Fragezeichen

Christian

franki29, 2021-05-06:
Hallo Christian,
bin leider kein Experte daher sieht mein Node ziemlich wild aus, da ich von meinem Internetradio die Infos abhole und dann an das Display schicke. Einige Sonderzeichen wie & oder ' sowie () habe ich bisher nicht hinbekommen.
Das eigentliche senden der Infos geht aber.

Ich nutze den http request note zum senden:

[{"id":"a3afeaa6.25db9","type":"http request","z":"d31b6ff4.b783c","name":"","method":"POST","ret":"txt","paytoqs":"ignore","url":"http://192.168.2.95/rest/api/v1/display/uid/19652/text?show={{payload}}","tls":"","persist":false,"proxy":"","authType":"basic","x":1450,"y":1340,"wires":[[]]}]

Dazu sende ich den Payload als String.

Was ist denn die Quelle der Daten?

franki29, 2021-05-06:
Anbei ein node zum senden von Daten eines MQTT Servers an das Display:

[{"id":"a9122d2a.035cb8","type":"http request","z":"d31b6ff4.b783c","name":"","method":"POST","ret":"txt","paytoqs":"ignore","url":"http://192.168.2.95/rest/api/v1/display/uid/18672/text?show= Aussen  Temp : {{payload.Aussentemperatur}} Feuchte : {{payload.AussenHumidity}} Druck : {{payload.Pressure}}","tls":"","persist":false,"proxy":"","authType":"basic","x":550,"y":580,"wires":[[]]},{"id":"4bd911a0.64b908","type":"mqtt in","z":"d31b6ff4.b783c","name":"","topic":"Sensor_aussen_7/#","qos":"2","datatype":"json","broker":"a3750f93.b119d","x":320,"y":580,"wires":[["a9122d2a.035cb8"]]},{"id":"a3750f93.b119d","type":"mqtt-broker","name":"Frankis","broker":"192.168.2.2","port":"1883","clientid":"nodered","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

wendtc, 2021-05-06:
Danke - hilft mir bestimmt 
Ich bin noch am Konfigurieren Teste Übertragen etc und versuche den aktuellen Slot auszulesen .
Ist bei dir im Web Interface etwas einstellbar ? bei mir geht nur bei installiertem Plugin das ändern der Texte oder Symbole.

franki29, 2021-05-06:
So weit ich das mitbekommen habe, muss man den Slot erst im Web konfigurieren und dann sieht man auch die UID.
Aber laut der Dokumentation https://github.com/BlueAndi/esp-rgb-led-matrix/blob/master/doc/WEBSOCKET.md
kann man die Slots auslesen. Habe ich aber nie benutzt.

wendtc, 2021-05-07:
Danke für die Inspirationen

BlueAndi, 2021-05-13:
FAQ extended: [How to change text properties](https://github.com/BlueAndi/esp-rgb-led-matrix/tree/PluginDevelopment#how-to-change-text-properties)