BlueAndi/Pixelix #45: Possible to add support for HUB75 Panels?

franki29, 2021-05-07:
Just a short question : Is it possible to add support for HUB75 Displays?
I have seen the https://github.com/pixelmatix/SmartMatrix library that also support ESP32
Would it be very difficulty to implement this?

Br. Frank

BlueAndi, 2021-05-07:
I don't have any HUB75 matrix panel, but it should be possible. Create a Board/Display class, which contains the hardware specific parts. See here e.g. https://github.com/BlueAndi/esp-rgb-led-matrix/issues/39#issuecomment-832928275

The PluginDevelopment branch currently supports switching between different boards/displays. See the commits from the last days.

Do you have such panels?

flcu, 2021-05-08:
Funny, I was looking at these a few days ago because of a kickstarter that's using them to build something similar to pixelix. I actually ordered two, so as soon as they are here I'll look at them.

franki29, 2021-05-08:
Hi  , thanks for the information. I have two of them at home, but I have absolute no programming skills, so I will be not able to integrate them. It would be great if flcu is able to do as the displays are cheap and have a nice plastic frame and, depending on the type, a smaller space between the LED's that would allow a sharper font.

flcu, 2021-05-08:
I will take a look, but it'll take some time.

Just be aware, that pixelix is right now only supporting a native resolution of 32x8 and these panels (the one I'll get) are 64x32. So the first version(s) will just scale the display two times, which means it'll be 64x16 and every "pixelix-pixel" will use a square of 4 LEDs. So it might even be "blockier" than sharper.

I know there's a branch for "dynamic display configuration" but I don't know the current status of that and if it tried to look nicer on bigger displays.

BlueAndi, 2021-05-10:
> Funny, I was looking at these a few days ago because of a kickstarter that's using them to build something similar to pixelix. I actually ordered two, so as soon as they are here I'll look at integrating them.

Can you share the link for the kickstarter project?

> Just be aware, that pixelix is right now only supporting a native resolution of 32x8 and these panels (the one I'll get) are 64x32.

Not 100% true :-) ... currently only a height of 8 pixels is supported, but the width may have more than 32 pixels. The plugins should always use the whole available width. 16 pixels height may be the next step, as it is a typical icon size. But there will be of course several further questions to answer, e.g. what about the right font, what about the number of lines to show, etc.

> So it might even be "blockier" than sharper.

The look will support the name "Pixelix". :-D

> I know there's a branch for "dynamic display configuration" but I don't know the current status of that and if it tried to look nicer on bigger displays.

The idea behind this branch was to configure the display topology during run time and not changing the sourcecode locally. But the branch is frozen for the moment.

> I will take a look, but it'll take some time.

Let's see what we can achieve.

BlueAndi, 2021-11-18:
Let's close the issue until there is the first HUB75 support available.

closed 2021-11-18 by BlueAndi

marianomd, 2023-10-26:
Hello! I'm going through it right now. Keep you posted.

[image]

marianomd, 2023-10-26:
I can't get past SSID screen. I tried AP mode and setting wifi credentials through serial terminal. 

```
        19 INFO             InitState.cpp:  485 PIXELIX starts up ...
        19 INFO             InitState.cpp:  486 Target: esp32doit-devkit-v1
        30 INFO             InitState.cpp:  487 SW version: v7.1.1:lc
        50 INFO             InitState.cpp:  489 SW revision: 2454575e59d70b5b51ec63b5f4856521662b1e11
        51 INFO             InitState.cpp:  490 ESP32 chip rev.: 3
        61 INFO             InitState.cpp:  491 ESP32 SDK version: v4.4.4
        82 INFO             InitState.cpp:  493 Wifi MAC: 94:E6:86:38:4C:04
        82 INFO             InitState.cpp:  494 LwIP version: 2.1.2esp
       116 DEBUG            ButtonDrv.cpp:  312 ButtonDrv task is ready.
       158 INFO    TopicHandlerService.cpp:   78 Topic handler service started.
       158 INFO       SettingsService.cpp:  328 Settings service started.
       410 INFO    SensorDataProvider.cpp:  132 Sensor GL5528: available
       410 INFO    SensorDataProvider.cpp:  132 Sensor SHT3x: -
       420 INFO    SensorDataProvider.cpp:  132 Sensor DHT11: -
       421 INFO    SensorDataProvider.cpp:  132 Sensor Battery: -
       439 INFO         PluginFactory.cpp:   81 Plugin type BTCQuotePlugin registered.
       440 INFO         PluginFactory.cpp:   81 Plugin type CountdownPlugin registered.
       450 INFO         PluginFactory.cpp:   81 Plugin type DateTimePlugin registered.
       460 INFO         PluginFactory.cpp:   81 Plugin type FirePlugin registered.
       460 INFO         PluginFactory.cpp:   81 Plugin type GithubPlugin registered.
       471 INFO         PluginFactory.cpp:   81 Plugin type IconTextLampPlugin registered.
       481 INFO         PluginFactory.cpp:   81 Plugin type IconTextPlugin registered.
       481 INFO         PluginFactory.cpp:   81 Plugin type JustTextPlugin registered.
       492 INFO         PluginFactory.cpp:   81 Plugin type MatrixPlugin registered.
       502 INFO         PluginFactory.cpp:   81 Plugin type OpenWeatherPlugin registered.
       512 INFO         PluginFactory.cpp:   81 Plugin type SunrisePlugin registered.
       512 INFO         PluginFactory.cpp:   81 Plugin type SysMsgPlugin registered.
       523 INFO         PluginFactory.cpp:   81 Plugin type ThreeIconPlugin registered.
       533 INFO         PluginFactory.cpp:   81 Plugin type WifiStatusPlugin registered.
       543 INFO         PluginFactory.cpp:   81 Plugin type WormPlugin registered.
       552 DEBUG       BrightnessCtrl.cpp:  294 Light: 0.000 (b-thr 0.000 < x < d-thr 0.000)
       553 DEBUG       BrightnessCtrl.cpp:  313 Change brightness goal to 25.
       565 DEBUG           DisplayMgr.cpp: 1223 ProcessTask is up.
       566 DEBUG           DisplayMgr.cpp: 1285 UpdateTask is up.
       576 INFO            DisplayMgr.cpp:  213 DisplayMgr is up.
       580 INFO            DisplayMgr.cpp:  326 Start plugin SysMsgPlugin (UID 39311) in slot 0.
       591 INFO            DisplayMgr.cpp:  341 Plugin SysMsgPlugin (UID 39311) installed in slot 0.
[   599][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
       599 INFO             InitState.cpp:  229 Clean up settings.
       679 INFO            DisplayMgr.cpp: 1112 Slot 0 (SysMsgPlugin) now active.
[  2647][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
      2802 INFO               APState.cpp:   95 Setup access point.
[  2809][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
[  2811][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ap_ssid NOT_FOUND
[  2819][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ap_passphrase NOT_FOUND
      3396 INFO               APState.cpp:  191 SSID: pixelix-86384c04
      3397 INFO              ResetMon.cpp:  108 Reset reason APP CPU: Software reset CPU
      3409 INFO              ResetMon.cpp:  109 Reset reason PRO CPU: Software reset CPU
      3409 INFO              ResetMon.cpp:  110 Counted resets: 1
```

Any ideas on what could be happening?

marianomd, 2023-10-26:
I got to the Hello World screen. Something is wrong, it seems the webserver responds slowly, the page loads incomplete, and finally after refreshing, the page doesn't load anymore.

When accessing the web interface, after logging in, I get these errors:

```
[328643][E][vfs_api.cpp:105] open(): /littlefs/style/bootstrap.min.css does not exist, no permits for creation
[329073][E][vfs_api.cpp:105] open(): /littlefs/js/menu.js.gz does not exist, no permits for creation
    356289 INFO                 Pages.cpp:  276 Invalid web request: /favicon.ico
[365112][E][vfs_api.cpp:105] open(): /littlefs/style/bootstrap.min.css does not exist, no permits for creation
[365314][E][vfs_api.cpp:105] open(): /littlefs/js/jquery-3.6.3.slim.min.js does not exist, no permits for creation
[384154][E][vfs_api.cpp:105] open(): /littlefs/style/bootstrap.min.css does not exist, no permits for creation
    400996 INFO                 Pages.cpp:  276 Invalid web request: /favicon.ico
    405881 INFO                 Pages.cpp:  276 Invalid web request: /favicon.ico
[408924][E][vfs_api.cpp:105] open(): /littlefs/style/bootstrap.min.css does not exist, no permits for creation
    604027 INFO              ResetMon.cpp:  108 Reset reason APP CPU: For APP CPU, reseted by PRO CPU
    604027 INFO              ResetMon.cpp:  109 Reset reason PRO CPU: Vbat power on reset
    604037 INFO              ResetMon.cpp:  110 Counted resets: 0
```

After some manual reboots I finally managed to see the menu, but some sluggishness is going on. Maybe out of RAM?

[image]

marianomd, 2023-10-26:
I reduced drastically the number of plugins, and managed to get the bootstrap css and menu loaded.

Still, I have reboots.

Got some stack overflows:

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (asyncTcpSock)
Core  1 register dump:
PC      : 0x400d58f7  PS      : 0x00060a36  A0      : 0x800e3e95  A1      : 0x3fff13a0  
A2      : 0x000000b8  A3      : 0x00000002  A4      : 0x00001000  A5      : 0x3ffc4e20
A6      : 0x00000080  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fff16c0  
A10     : 0x80808081  A11     : 0x3fff13e0  A12     : 0x00000002  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008ab11  LEND    : 0x4008ab35  LCOUNT  : 0x80101d58

Backtrace: 0x400d58f4:0x3fff13a0 0x400e3e92:0x3fff13e0 0x400e3196:0x3fff5440 0x400e32fe:0x3fff54a0 0x400e3361:0x3fff54d0 0x401a854f:0x3fff54f0 0x401a8952:0x3fff5510 0x400ea19a:0x3fff5560 0x400ea1c1:0x3fff55a0 0x40194005:0x3fff55c0 0x4019375a:0x3fff55f0

  #0  0x400d58f4:0x3fff13a0 in Rgb888::operator unsigned int() const at lib/YAGfx/src/Rgb888.h:179
      (inlined by) DisplayMgr::getFBCopy(unsigned int*, unsigned int, unsigned char*) at src/Gfx/DisplayMgr.cpp:685
  #1  0x400e3e92:0x3fff13e0 in WsCmdGetDisp::execute(AsyncWebSocket*, AsyncWebSocketClient*) at src/Web/WsCommand/WsCmdGetDisp.cpp:87
  #2  0x400e3196:0x3fff5440 in WebSocketSrv::handleMsg(AsyncWebSocket*, AsyncWebSocketClient*, char const*, unsigned int) at src/Web/WebSocket.cpp:379      
  #3  0x400e32fe:0x3fff54a0 in WebSocketSrv::onData(AsyncWebSocket*, AsyncWebSocketClient*, AwsFrameInfo*, unsigned char const*, unsigned int) at src/Web/WebSocket.cpp:290
  #4  0x400e3361:0x3fff54d0 in WebSocketSrv::onEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at src/Web/WebSocket.cpp:216
      (inlined by) WebSocketSrv::onEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at src/Web/WebSocket.cpp:184
  #5  0x401a854f:0x3fff54f0 in std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), void (*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::_M_invoke(std::_Any_data const&, AsyncWebSocket*&&, AsyncWebSocketClient*&&, AwsEventType&&, void*&&, unsigned char*&&, unsigned int&&) at c:\users\user\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #6  0x401a8952:0x3fff5510 in std::function<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::operator()(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) const at c:\users\user\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncWebSocket::_handleEvent(AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at .pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/AsyncWebSocket.cpp:867
  #7  0x400ea19a:0x3fff5560 in AsyncWebSocketClient::_onData(void*, unsigned int) at .pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/AsyncWebSocket.cpp:685
  #8  0x400ea1c1:0x3fff55a0 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, void*, unsigned int)#7}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/AsyncWebSocket.cpp:484
      (inlined by) _M_invoke at c:\users\user\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297        
  #9  0x40194005:0x3fff55c0 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\users\user\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncClient::_sockIsReadable() at .pio/libdeps/esp32doit-devkit-v1/AsyncTCPSock/src/AsyncTCP.cpp:885
  #10 0x4019375a:0x3fff55f0 in _asynctcpsock_task(void*) at .pio/libdeps/esp32doit-devkit-v1/AsyncTCPSock/src/AsyncTCP.cpp:168

```

```

***ERROR*** A stack overflow in task asyncTcpSock has been detected.

Backtrace: 0x40083f2d:0x3fff1130 0x4008e95d:0x3fff1150 0x40092011:0x3fff1170 0x40090513:0x3fff11f0 0x4008ea6c:0x3fff1220 0x4008ea1c:0xabba1234 |<-CORRUPTED

  #0  0x40083f2d:0x3fff1130 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008e95d:0x3fff1150 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40092011:0x3fff1170 in vApplicationStackOverflowHook at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:407
  #3  0x40090513:0x3fff11f0 in vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3504
  #4  0x4008ea6c:0x3fff1220 in _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
[... cut ...]
  #11 0x40194005:0x3fff55b0 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\users\user\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncClient::_sockIsReadable() at .pio/libdeps/esp32doit-devkit-v1/AsyncTCPSock/src/AsyncTCP.cpp:885
  #12 0x4019375a:0x3fff55e0 in _asynctcpsock_task(void*) at .pio/libdeps/esp32doit-devkit-v1/AsyncTCPSock/src/AsyncTCP.cpp:168
```

Heap usage seems fine:

[image]

Any help will be greatly appreciated.

marianomd, 2023-10-26:
Well I managed to make it work. I had to disable the websocket display preview, as it's too large for 64x64 pixels.

Here's the code if someone wants it: https://github.com/marianomd/esp-rgb-led-matrix

BlueAndi, 2023-10-26:
@marianomd good job and thanks for providing it. Do you plan to update the plugins too, to support 64x64 pixels?
I will have a look to your code.

marianomd, 2023-10-27:
It's a possibility yes! I want to thank you for the amount of work this project has. I like the architecture/abstraction layer very much. I have made an adaptation attempt with awtrix-light too, but lots of coupling.

Also adapted [EspHoMaTriXv2](https://github.com/marianomd/EspHoMaTriXv2-HUB75), to work with esphome.

For the time being I designed what I needed with esphome directly in the yaml:

[image]

marianomd, 2023-10-27:
Regarding my code, it's a bunch of quick hacks, like modifying an existing target in the ini, because I didn't know how to address the two variants or commenting out the preview image generator.

BlueAndi, 2023-10-29:
> It's a possibility yes! In fact, several of them work right out the box. I want to thank you for the amount of work this project has. I like the architecture/abstraction layer very much. I have made an adaptation attempt with awtrix-light too, but lots of coupling.

Thanks, thats sound for the heart. :-)

The plugin should scale automatically on the x-axis, so thats what you saw. On the y-axis its more difficult and the original display has only 8 pixel height anway. So I took only care about them. Having plugins supporting this, will also increase code size and at the moment the 4 MB flash modules are already limiting the features. With 8 MB or more flash it looks much better.

BlueAndi, 2023-12-29:
@marianomd I added the HUB75 support to the Development branch. Can you verify it, I still have no panel by myself? Thanks again for sharing!

marianomd, 2024-01-11:
Hello Andreas! This is great! I have the clock in use by the family, but I will find a moment to test it.