BlueAndi/Pixelix #49: can't build

filmferry, 2021-05-30:
Verbose mode can be enabled via `-v, --verbose` option
fatal: not a git repository (or any of the parent directories): .git
CalledProcessError: Command '['git', 'rev-parse', '--abbrev-ref', 'HEAD']' returned non-zero exit status 128.:
  File "C:\Users\gokul\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\gokul\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\gokul\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\gokul\Pictures\esp 3.0.0\scripts\getGitRev.py", line 84:
    softwareVersion         = "-DSW_VERSION=" + getVersion()
  File "C:\Users\gokul\Pictures\esp 3.0.0\scripts\getGitRev.py", line 63:
    branchName = getCurrentGitBranch()
  File "C:\Users\gokul\Pictures\esp 3.0.0\scripts\getGitRev.py", line 35:
    return subprocess.check_output(['git', 'rev-parse', '--abbrev-ref', 'HEAD']).decode('ascii').strip()
  File "C:\Users\gokul\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\gokul\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)
================================================================= [FAILED] Took 1.69 seconds =================================================================

Environment              Status    Duration
-----------------------  --------  ------------
esp32doit-devkit-v1-usb  FAILED    00:00:01.685
=========================================================== 1 failed, 0 succeeded in 00:00:01.685 ===========================================================
The terminal process "C:\Users\gokul\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

0ooYoo0, 2021-05-30:
Please have a look at #47.

BlueAndi, 2021-06-06:
@filmferry Any progress?

BlueAndi, 2021-06-10:
I close the issue. Please reopen if necessary.

closed 2021-06-10 by BlueAndi