BlueAndi/Pixelix #50: Version 3 display is not smooth

PabluskiNC, 2021-06-08:
I have compiled the new version. Complements on the OpenWeather and other updates.
However, the actual LED display is not as smooth as it was in version 2.
I have re-loaded the version 2 for now - I will be creating a test display to compare the two versions.
    
ESP type: | ESP32
ESP Chip Rev.: | 1
ESP Chip ID: | 749C23286F24
ESP CPU Freq.: | 240 MHz
Flash Chip Mode: | DIO
Flash Chip Size : | 4 MByte
Flash Chip Speed: | 40 MHz

Version: | 271145f8f3ca442aaa7972e9c84f26388f53e253
-- | --
ESP SDK Version: | v3.2.3-14-gd3e562907
Arduino IDF Version: | release/v3.2
LwIP Version: | 2.0.3

PabluskiNC, 2021-06-09:
More info - I am using the JustText plugin for some scrolling text - I took videos of the two displays ...  but the files are too big for this.

BlueAndi, 2021-06-09:
Can you share via google drive or any other platform?

PabluskiNC, 2021-06-09:
https://photos.app.goo.gl/hdzAmXdmDY4oRoHb9 the one with the wooden background is the V3 test setup.

BlueAndi, 2021-06-09:
Can you please check the text scroll pause in the settings page?
Is it really the same value in both versions?

PabluskiNC, 2021-06-10:
V2 was at 20 ms; but I'm not sure about the V3 - I shot a new video with it at 20 ms. 
It is better, but still seems slow compared to version 2.
it is in the google folder.

BlueAndi, 2021-06-10:
20ms are always challenging ;-), because scrollling takes place in the same time period, than updating the physical display.
Typically a display update (32x8) itself needs about 8ms.

Can you switch to this branch? https://github.com/BlueAndi/esp-rgb-led-matrix/tree/v3.1.x
It is equal to v3, but contains some statistic timing outputs via logging. I am interesting regarding the values.

BlueAndi, 2021-06-10:
BTW with version 2 you mean v2.2.0, right?

PabluskiNC, 2021-06-10:
I have created a second test display using the same esp32 hardware (HW-607 ESP-WROOM-32) - I captured both at the same time - topis V3 bottom is V2 - both are set to 20ms.

PabluskiNC, 2021-06-10:
Loaded 3.1.x
|112239| INFO: DisplayMgr.cpp:1142 [ 19, 20, 21 ]
|112240| INFO: DisplayMgr.cpp:1154 [  2,  2,  6 ] [  7,  8,  8 ] [ 10, 10, 14 ]
|114951| INFO: DisplayMgr.cpp:1003 Slot 1 (DateTimePlugin) now active.
|116240| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|116240| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|120240| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|120240| INFO: DisplayMgr.cpp:1154 [  1,  1,  2 ] [  8,  8,  8 ] [  9,  9, 10 ]
|124243| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|124243| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|128246| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|128246| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|132249| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|132250| INFO: DisplayMgr.cpp:1154 [  0,  1,  2 ] [  8,  8,  8 ] [  8,  9, 10 ]
|136251| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|136251| INFO: DisplayMgr.cpp:1154 [  0,  1,  2 ] [  8,  8,  8 ] [  8,  9, 10 ]
|140254| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|140254| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|144257| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|144257| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|144950| INFO: DisplayMgr.cpp:1003 Slot 2 (JustTextPlugin) now active.
|148257| INFO: DisplayMgr.cpp:1142 [ 19, 20, 21 ]
|148258| INFO: DisplayMgr.cpp:1154 [  0,  2,  7 ] [  7,  8,  8 ] [  8, 10, 15 ]
|152258| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|152259| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|155484| INFO: WebSocket.cpp:220 ws[/ws][1] Client connected.
|156258| INFO: DisplayMgr.cpp:1142 [ 19, 20, 21 ]
|156259| INFO: DisplayMgr.cpp:1154 [  2,  2,  3 ] [  8,  8,  8 ] [ 10, 10, 11 ]
|160259| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|160260| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]

BlueAndi, 2021-06-10:
Looks good, the update task period is about 20ms. Thats fine.
And just to sure, if you set v3.1.x to 20ms, you restart Pixelix, right? Because for taking effect, it must be restarted.

PabluskiNC, 2021-06-10:
|29600| INFO: ClockDrv.cpp:109 Time successfully synchronized: 16:24
|29600| INFO: ConnectedState.cpp:118 Hostname: pixelix IP: 192.168.170.112
|29610| INFO: DisplayMgr.cpp:1003 Slot 0 (SysMsgPlugin) now active.
|32181| INFO: DisplayMgr.cpp:1142 [ 19, 20, 21 ]
|32182| INFO: DisplayMgr.cpp:1154 [  0,  2, 12 ] [  8,  8,  8 ] [  8, 10, 20 ]
|33082| INFO: PluginMgr.cpp:514 JSON document size: 32
|36182| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|36183| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|40183| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|40184| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|44184| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|44185| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|48185| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|48186| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|52186| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|52187| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|54838| INFO: DisplayMgr.cpp:1003 Slot 1 (DateTimePlugin) now active.
|56187| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|56188| INFO: DisplayMgr.cpp:1154 [  2,  2,  3 ] [  8,  8,  8 ] [ 10, 10, 11 ]
|60187| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|60187| INFO: DisplayMgr.cpp:1154 [  1,  1,  2 ] [  8,  8,  8 ] [  9,  9, 10 ]
|64189| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|64189| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|68191| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|68191| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|72193| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|72194| INFO: DisplayMgr.cpp:1154 [  0,  1,  2 ] [  8,  8,  8 ] [  8,  9, 10 ]
|76194| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|76194| INFO: DisplayMgr.cpp:1154 [  1,  1,  2 ] [  8,  8,  8 ] [  9,  9, 10 ]
|80196| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|80196| INFO: DisplayMgr.cpp:1154 [  0,  1,  1 ] [  8,  8,  8 ] [  8,  9,  9 ]
|84197| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|84197| INFO: DisplayMgr.cpp:1154 [  1,  1,  2 ] [  8,  8,  8 ] [  9,  9, 10 ]
|84849| INFO: DisplayMgr.cpp:1003 Slot 2 (JustTextPlugin) now active.
|88199| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|88200| INFO: DisplayMgr.cpp:1154 [  1,  2,  3 ] [  8,  8,  8 ] [  9, 10, 11 ]
|92200| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|92201| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|96201| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|96202| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|100202| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|100203| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|104222| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|104223| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|108223| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|108224| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|112224| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|112225| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]

PabluskiNC, 2021-06-10:
Yes, restarted on both traces.

BlueAndi, 2021-06-10:
Hmm ... I am on track ... may need a little bit.

BlueAndi, 2021-06-10:
Please pull the branch and try again.

PabluskiNC, 2021-06-10:
Tried again - much better! 
|119397| INFO: DisplayMgr.cpp:1003 Slot 2 (JustTextPlugin) now active.
|120427| INFO: DisplayMgr.cpp:1142 [ 20, 20, 21 ]
|120428| INFO: DisplayMgr.cpp:1154 [  0,  2,  3 ] [  8,  8,  8 ] [  8, 10, 11 ]
|124447| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|124448| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|128467| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|128468| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|132487| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|132488| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|136507| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|136508| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|140527| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|140528| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|144547| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|144548| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]
|148567| INFO: DisplayMgr.cpp:1142 [ 20, 20, 20 ]
|148568| INFO: DisplayMgr.cpp:1154 [  2,  2,  2 ] [  8,  8,  8 ] [ 10, 10, 10 ]

BlueAndi, 2021-06-10:
Ok, perfect. I will bump a v3.0.1 in the next minutes. The bugfix version will have the statistics disabled.

BlueAndi, 2021-06-10:
BTW good finding! Thanks!

closed 2021-06-11 by BlueAndi