BlueAndi/Pixelix #51: Cant build in platformio

zx804262156, 2021-06-09:
> Executing task: C:\Users\zhaoxin\.platformio\penv\Scripts\platformio.exe run <

Traceback (most recent call last):
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\core.py", line 1256, in invoke
    Command.invoke(self, ctx)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\__main__.py", line 63, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 48, in on_platformio_start     
    telemetry.on_command()
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 292, in on_command
    resend_backuped_reports()
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 443, in resend_backuped_reports  
    tm = app.get_state_item("telemetry", {})
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\app.py", line 191, in get_state_item
    with State() as state:
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\app.py", line 94, in __init__
    self.path = join(get_project_core_dir(), "appstate.json")
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\helpers.py", line 51, in get_project_core_dir
    join(get_project_dir(), "platformio.ini")
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 441, in get_instance
    instance = {"mtime": mtime, "config": ProjectConfig(path)}
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 102, in __init__
    self.read(path, parse_extra)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 114, in read
    self._parser.read(path)
  File "C:\Users\zhaoxin\.platformio\python3\lib\configparser.py", line 696, in read
    self._read(fp, filename)
  File "C:\Users\zhaoxin\.platformio\python3\lib\configparser.py", line 1014, in _read
    for lineno, line in enumerate(fp, start=1):
UnicodeDecodeError: 'gbk' codec can't decode byte 0xa6 in position 2056: illegal multibyte sequence

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zhaoxin\.platformio\penv\Scripts\platformio-script.py", line 11, in <module>
    load_entry_point('platformio==5.1.1', 'console_scripts', 'platformio')()
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\__main__.py", line 115, in main
    maintenance.on_platformio_exception(e)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 77, in on_platformio_exception
    telemetry.on_exception(e)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 322, in on_exception
    send_exception(description, is_fatal)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 394, in send_exception
    mp = MeasurementProtocol()
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 74, in __init__
    self["cid"] = app.get_cid()
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\app.py", line 256, in get_cid
    cid = get_state_item("cid")
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\app.py", line 191, in get_state_item
    with State() as state:
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\app.py", line 94, in __init__
    self.path = join(get_project_core_dir(), "appstate.json")
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\helpers.py", line 51, in get_project_core_dir
    join(get_project_dir(), "platformio.ini")
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 441, in get_instance
    instance = {"mtime": mtime, "config": ProjectConfig(path)}
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 102, in __init__
    self.read(path, parse_extra)
  File "C:\Users\zhaoxin\.platformio\penv\lib\site-packages\platformio\project\config.py", line 114, in read
    self._parser.read(path)
  File "C:\Users\zhaoxin\.platformio\python3\lib\configparser.py", line 696, in read
    self._read(fp, filename)
  File "C:\Users\zhaoxin\.platformio\python3\lib\configparser.py", line 1014, in _read
    for lineno, line in enumerate(fp, start=1):
UnicodeDecodeError: 'gbk' codec can't decode byte 0xa6 in position 2056: illegal multibyte sequence
Terminal process“C:\Users\zhaoxin\.platformio\penv\Scripts\platformio.exe 'run'”Terminated, exit code: 1。

The terminal will be reused by the task, press any key to close it.

BlueAndi, 2021-06-09:
1. Can you uninstall all python interpreters from system?
2. And restart VSCode.

Do you have the latest pio version?
Did you any change in the platform.ini?

zx804262156, 2021-06-09:
@BlueAndi 

I deleted all the comments in platformio.ini and it's ok.
However, there is a problem with the startup, and it still cannot be replaced with ESP32

--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---  
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
|40| INFO: InitState.cpp:335 PIXELIX starts up ...
|41| INFO: InitState.cpp:336 SW version: v3.0.0:lc
|41| INFO: InitState.cpp:337 SW revision: 4b2b2ac22e40b4ee8b41ee60c1725e4f5634e41a
|46| INFO: InitState.cpp:338 ESP32 chip rev.: 1
|50| INFO: InitState.cpp:339 ESP32 SDK version: v3.3.5-1-g85c43024c
|57| INFO: InitState.cpp:340 Ambient light sensor detected: 0
|62| INFO: InitState.cpp:341 Wifi MAC: 84:CC:A8:7A:BF:B0
|67| INFO: InitState.cpp:342 LwIP version: 2.0.3
|72| INFO: ButtonDrv.cpp:153 ButtonDrv task is ready.
|73| ERROR: SPIFFS:0 mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
A[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
|96| INFO: wifi:0 : mount failed, -10025
|0| INFO: :0
|0| INFO: :0 iver task: 3ffb55dc, prio:23, stack:3584, core=0
|256| INFO: wifi:0 sk: 3ffb55dc, prio:23, stack:3584, core=0
|0| INFO: :0 3, stack:3584, core=0
|0| INFO: :0 rmware version: dc30037
|258| INFO: wifi:0 version: dc30037
|0| INFO: :0 7
|0| INFO: :0 NVS flash: enabled
|265| INFO: wifi:0 sh: enabled
|0| INFO: :0 , stack:3584, core=0
|0| INFO: :0 nano formating: disabled
|275| INFO: wifi:0 rmating: disabled
|0| INFO: :0 84, core=0
|0| INFO: :0 ta frame dynamic rx buffer num: 32
|284| INFO: wifi:0 e dynamic rx buffer num: 32
|0| INFO: :0 re=0
|0| INFO: :0 nagement frame dynamic rx buffer num: 32
|295| INFO: wifi:0 t frame dynamic rx buffer num: 32
|0| INFO: :0 num: 32
|0| INFO: :0 nagement short buffer num: 32
|305| INFO: wifi:0 t short buffer num: 32
|0| INFO: :0 32
|0| INFO: :0 namic tx buffer num: 32
|314| INFO: wifi:0 x buffer num: 32
|0| INFO: :0
|0| INFO: :0 atic rx buffer size: 1600
|322| INFO: wifi:0  buffer size: 1600
|0| INFO: :0  32
|0| INFO: :0 atic rx buffer num: 16
|330| INFO: wifi:0  buffer num: 16
|0| INFO: :0 32
|0| INFO: :0 namic rx buffer num: 32
|427| INFO: wifi:0 x buffer num: 32
|0| INFO: :0  32
|0| INFO: :0 sta (84:cc:a8:7a:bf:b0)
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|957| WARNING: PluginMgr.cpp:150 Plugin installation is empty.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|969| INFO: ErrorState.cpp:68 Going in error state.

--- exit ---

zx804262156, 2021-06-09:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

What does this error mean

BlueAndi, 2021-06-09:
It seems that the filesystem was not programmed, therefore it couldn't be mounted.

BlueAndi, 2021-06-10:
@zx804262156 Can you revert your platformio.ini changes and only remove these two comments?
```
; Parses ALL C/C++ source files of the project and follows only by nested includes (#include ..., chain…) from the libraries.
; And evaluates C/C++ Preprocessor conditional syntax.
```

Your provided trace looks like one of them may cause the problem. I assume you changed nothing else in the file.

zx804262156, 2021-06-11:
@BlueAndi Modify the file "platformio.ini" according to your method successfully!
But it can't run now, I can't solve it!
The following is the error code:

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
|41| INFO: InitState.cpp:335 PIXELIX starts up ...
|41| INFO: InitState.cpp:336 SW version: v3.0.0:lc
|42| INFO: InitState.cpp:337 SW revision: 4b2b2ac22e40b4ee8b41ee60c1725e4f5634e41a
|47| INFO: InitState.cpp:338 ESP32 chip rev.: 1
|51| INFO: InitState.cpp:339 ESP32 SDK version: v3.3.5-1-g85c43024c
|58| INFO: InitState.cpp:340 Ambient light sensor detected: 1
|63| INFO: InitState.cpp:341 Wifi MAC: 84:CC:A8:7A:BF:B0
|67| INFO: InitState.cpp:342 LwIP version: 2.0.3
|73| INFO: ButtonDrv.cpp:153 ButtonDrv task is ready.
|74| ERROR: SPIFFS:0 mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
t[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND   
|97| INFO: wifi:0 : mount failed, -10025
|0| INFO: :0
|0| INFO: :0 iver task: 3ffb55dc, prio:23, stack:3584, core=0
|257| INFO: wifi:0 sk: 3ffb55dc, prio:23, stack:3584, core=0
|0| INFO: :0 3, stack:3584, core=0
|0| INFO: :0 rmware version: dc30037
|258| INFO: wifi:0 version: dc30037
|0| INFO: :0 7
|0| INFO: :0 NVS flash: enabled
|266| INFO: wifi:0 sh: enabled
|0| INFO: :0 , stack:3584, core=0
|0| INFO: :0 nano formating: disabled
|275| INFO: wifi:0 rmating: disabled
|0| INFO: :0 84, core=0
|0| INFO: :0 ta frame dynamic rx buffer num: 32
|285| INFO: wifi:0 e dynamic rx buffer num: 32
|0| INFO: :0 re=0
|0| INFO: :0 nagement frame dynamic rx buffer num: 32
|295| INFO: wifi:0 t frame dynamic rx buffer num: 32
|0| INFO: :0 num: 32
|0| INFO: :0 nagement short buffer num: 32
|306| INFO: wifi:0 t short buffer num: 32
|0| INFO: :0 32
|0| INFO: :0 namic tx buffer num: 32
|315| INFO: wifi:0 x buffer num: 32
|0| INFO: :0
|0| INFO: :0 atic rx buffer size: 1600
|323| INFO: wifi:0  buffer size: 1600
|0| INFO: :0  32
|0| INFO: :0 atic rx buffer num: 16
|331| INFO: wifi:0  buffer num: 16
|0| INFO: :0 32
|0| INFO: :0 namic rx buffer num: 32
|428| INFO: wifi:0 x buffer num: 32
|0| INFO: :0  32
|0| INFO: :0 sta (84:cc:a8:7a:bf:b0)
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|958| WARNING: PluginMgr.cpp:150 Plugin installation is empty.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|970| INFO: ErrorState.cpp:68 Going in error state.
ets Jun  8 2016 00:22:57

[... 45 lines cut ...]
|0| INFO: :0 num: 32
|0| INFO: :0 nagement short buffer num: 32
|306| INFO: wifi:0 t short buffer num: 32
|0| INFO: :0 32
|0| INFO: :0 namic tx buffer num: 32
|315| INFO: wifi:0 x buffer num: 32
|0| INFO: :0
|0| INFO: :0 atic rx buffer size: 1600
|323| INFO: wifi:0  buffer size: 1600
|0| INFO: :0  32
|0| INFO: :0 atic rx buffer num: 16
|331| INFO: wifi:0  buffer num: 16
|0| INFO: :0 32
|0| INFO: :0 namic rx buffer num: 32
|427| INFO: wifi:0 x buffer num: 32
|0| INFO: :0  32
|0| INFO: :0 sta (84:cc:a8:7a:bf:b0)
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|957| WARNING: PluginMgr.cpp:150 Plugin installation is empty.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: plugin_install NOT_FOUND
|969| INFO: ErrorState.cpp:68 Going in error state.

BlueAndi, 2021-06-11:
You need to program the filesystem:
[image]

BlueAndi, 2021-06-11:
v3.0.2 released, please try. This time it should't be necessary to remove something from platformio.ini

closed 2021-07-14 by BlueAndi