BlueAndi/Pixelix #56: Always starts in AP Mode

SGiehler, 2021-07-07:
Hi,

i managed to uploiad the firmware aswell as the FS, but when i save a SSID / Passphrase via the UI nothing happens. It does report "ok" in an alert box, but when i reboot it still fires up "pixelix" SSID. Any idea how i can solve this?

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
        32 INFO             InitState.cpp:  349 PIXELIX starts up ...
        32 INFO             InitState.cpp:  350 SW version: v3.0.2:lc:b
        33 INFO             InitState.cpp:  351 SW revision: e2b8f75251d30e5f517d337ad6445686ca3fd4d4
        43 INFO             InitState.cpp:  352 ESP32 chip rev.: 1
        48 INFO             InitState.cpp:  353 ESP32 SDK version: v3.3.5-1-g85c43024c
        56 INFO             InitState.cpp:  354 Wifi MAC: 9C:9C:1F:E2:28:1C
        62 INFO             InitState.cpp:  355 LwIP version: 2.0.3
        70 INFO             ButtonDrv.cpp:  153 ButtonDrv task is ready.
       299 INFO    SensorDataProvider.cpp:   80 Sensor GL5528: -
       299 INFO    SensorDataProvider.cpp:   80 Sensor SHT3x: -
       299 INFO    SensorDataProvider.cpp:   80 Sensor DHT11: -
       492 INFO         PluginFactory.cpp:   81 Plugin type BTCQuotePlugin registered.
       492 INFO         PluginFactory.cpp:   81 Plugin type CountdownPlugin registered.
       496 INFO         PluginFactory.cpp:   81 Plugin type DatePlugin registered.
       503 INFO         PluginFactory.cpp:   81 Plugin type DateTimePlugin registered.
       511 INFO         PluginFactory.cpp:   81 Plugin type FirePlugin registered.
       518 INFO         PluginFactory.cpp:   81 Plugin type GameOfLifePlugin registered.
       526 INFO         PluginFactory.cpp:   81 Plugin type GruenbeckPlugin registered.
       533 INFO         PluginFactory.cpp:   81 Plugin type IconTextLampPlugin registered.
       541 INFO         PluginFactory.cpp:   81 Plugin type IconTextPlugin registered.
       549 INFO         PluginFactory.cpp:   81 Plugin type JustTextPlugin registered.
       556 INFO         PluginFactory.cpp:   81 Plugin type OpenWeatherPlugin registered.
       564 INFO         PluginFactory.cpp:   81 Plugin type RainbowPlugin registered.
       572 INFO         PluginFactory.cpp:   81 Plugin type SensorPlugin registered.
       579 INFO         PluginFactory.cpp:   81 Plugin type ShellyPlugSPlugin registered.
       587 INFO         PluginFactory.cpp:   81 Plugin type SunrisePlugin registered.
       594 INFO         PluginFactory.cpp:   81 Plugin type SysMsgPlugin registered.
       602 INFO         PluginFactory.cpp:   81 Plugin type TempHumidPlugin registered.
       609 INFO         PluginFactory.cpp:   81 Plugin type TestPlugin registered.
       616 INFO         PluginFactory.cpp:   81 Plugin type TimePlugin registered.
       624 INFO         PluginFactory.cpp:   81 Plugin type VolumioPlugin registered.
       631 INFO         PluginFactory.cpp:   81 Plugin type WifiStatusPlugin registered.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: slot_cfg NOT_FOUND
       646 WARNING         DisplayMgr.cpp: 1156 Display slot configuration is empty.
       654 INFO            DisplayMgr.cpp:  202 DisplayMgr is up.
       664 INFO            DisplayMgr.cpp:  330 Plugin SysMsgPlugin installed in slot 0.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
       693 INFO            DisplayMgr.cpp:  952 Slot 0 (SysMsgPlugin) now active.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
     22547 INFO                      wifi:    0 
         0 INFO                          :    0
         0 INFO                          :    0 iver task: 3ffafd78, prio:23, stack:3584, core=0
     27278 INFO                      wifi:    0 : 3ffafd78, prio:23, stack:3584, core=0
         0 INFO                          :    0 3, stack:3584, core=0
         0 INFO                          :    0 rmware version: dc30037
     27288 INFO                      wifi:    0 rsion: dc30037
         0 INFO                          :    0 7
         0 INFO                          :    0 NVS flash: enabled
     27303 INFO                      wifi:    0 : enabled
         0 INFO                          :    0 , stack:3584, core=0
         0 INFO                          :    0 nano formating: disabled
     27321 INFO                      wifi:    0 ating: disabled
         0 INFO                          :    0 84, core=0
         0 INFO                          :    0 ta frame dynamic rx buffer num: 32
     27339 INFO                      wifi:    0 dynamic rx buffer num: 32
         0 INFO                          :    0 re=0
         0 INFO                          :    0 nagement frame dynamic rx buffer num: 32
     27357 INFO                      wifi:    0 frame dynamic rx buffer num: 32
         0 INFO                          :    0 num: 32
         0 INFO                          :    0 nagement short buffer num: 32
     27376 INFO                      wifi:    0 short buffer num: 32
         0 INFO                          :    0 32
         0 INFO                          :    0 namic tx buffer num: 32
     27393 INFO                      wifi:    0 buffer num: 32
         0 INFO                          :    0
         0 INFO                          :    0 atic rx buffer size: 1600
     27409 INFO                      wifi:    0 uffer size: 1600
         0 INFO                          :    0  32
         0 INFO                          :    0 atic rx buffer num: 16
     27425 INFO                      wifi:    0 uffer num: 16
         0 INFO                          :    0 32
         0 INFO                          :    0 namic rx buffer num: 32
     27540 INFO                      wifi:    0 buffer num: 32
         0 INFO                          :    0
         0 INFO                          :    0 softAP (9c:9c:1f:e2:28:1d)
     27546 INFO                      wifi:    0 c:9c:1f:e2:28:1d)
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 ower save buffer number: 16
     27563 INFO                      wifi:    0  buffer number: 16
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 x length of beacon: 752/752
     27581 INFO                      wifi:    0 of beacon: 752/752
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 x length of beacon: 752/752
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
     27615 INFO               APState.cpp:   93 Setup access point.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: ap_ssid NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: ap_passphrase NOT_FOUND
     28391 INFO                      wifi:    0 of beacon: 752/752
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 ower save buffer number: 16
     28406 INFO               APState.cpp:  186 SSID: pixelix
     28414 INFO            DisplayMgr.cpp:  952 Slot 0 (SysMsgPlugin) now active.`

BlueAndi, 2021-07-07:
Hi Stefan, do you have a user button connected, which is pressed during startup? This forces to start the access point mode.

SGiehler, 2021-07-07:
Hi Andi, now i'm feeling kinda dumb. I indeed had my switch in a wrong oriantation. 👍

closed 2021-07-07 by SGiehler