BlueAndi/Pixelix #58: Custom plugin not detected

ReinVelt, 2021-09-26:
I have created my own plugin following yours rules in plugins_dev.md . The files are compiled successfully and uploaded but the plugin is not working because it is not detected. PluginFactory.cpp detects all other plugins but not mine.

So here is my dumb question: How does pluginFactory get the list with available plugins?  Should i add something to a config file?

BlueAndi, 2021-09-26:
Register the plugin in the InitState here:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/0c0b6bbee25cfb1dbdeda7b3249d0a51e9eba9ca/src/StateMachine/InitState.cpp#L379

I will add this to the documentation.

ReinVelt, 2021-09-27:
Yes, it works. Thank you very much for your support.

closed 2021-09-27 by ReinVelt