BlueAndi/Pixelix #63: Consider a sound spectrum visualization plugin

dragospascale, 2021-12-07:
It will be a nice feature to have. Maybe using a microphone . Something like ESP32_FFT_VU project or Sound Reactive WLED.
Your idea of plugins is better than anyone else

BlueAndi, 2021-12-08:
Thanks for the idea. Currently I have no microphone available, but maybe in the future.
One question will be how many time a FFT needs covering 500 Hz to 20kHz.

dragospascale, 2021-12-09:
Look at the code at https://github.com/s-marley/ESP32-INMP441-Matrix-VU.
I'm using it on a flexible led matrix 32x8 ws2812B and it works great

BlueAndi, 2021-12-22:
I2S interface:
* Word Select (WS) - 15
* Serial Data (SD) - 32
* Serial Clock (SCK) - 14

Microphone interface:
* Left/Right Clock (LRCL)
* Data Output (DOUT)
* Bit Clock (BCLK)

Connection:
* WS -- LRCL
* SD -- DOUT
* SCK -- SCK

BlueAndi, 2022-01-19:
Note, the I2S interface will change with Arduino 2.x, because the underlying IDF version changed it. The upper mentioned projects are using currently deprecated functionality, which may not be supported in the new Arduino version.

BlueAndi, 2022-02-18:
SoundReactivePlugin is ready for testing, after a lot of hours.
Current drawback, the 4M flash is nearly full.

closed 2022-02-18 by BlueAndi