BlueAndi/Pixelix #65: captive portal does not start

kuifje123, 2022-01-14:
Hi Andi,

I am using the pixelix on an ESP32 from AZ Delivery.
Building and programming went well. I then start up the device with the user button pressed, I see the access point and can connect to that. However then it stops, the captive portal is not showing up.
Tried this both on a macbook and on iPhone.

Here is some logging I see:

`[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
      7859 INFO                      wifi:    0 
         0 INFO                          :    0 
         0 INFO                          :    0 iver task: 3ffcbbb8, prio:23, stack:3584, core=0
     15966 INFO                      wifi:    0 : 3ffcbbb8, prio:23, stack:3584, core=0
         0 INFO                          :    0 3, stack:3584, core=0
         0 INFO                          :    0 rmware version: dc30037
     15976 INFO                      wifi:    0 rsion: dc30037
         0 INFO                          :    0 7
         0 INFO                          :    0 NVS flash: enabled
     15991 INFO                      wifi:    0 : enabled
         0 INFO                          :    0 , stack:3584, core=0
         0 INFO                          :    0 nano formating: disabled
     16009 INFO                      wifi:    0 ating: disabled
         0 INFO                          :    0 84, core=0
         0 INFO                          :    0 ta frame dynamic rx buffer num: 32
     16027 INFO                      wifi:    0 dynamic rx buffer num: 32
         0 INFO                          :    0 re=0
         0 INFO                          :    0 nagement frame dynamic rx buffer num: 32
     16045 INFO                      wifi:    0 frame dynamic rx buffer num: 32
         0 INFO                          :    0 num: 32
         0 INFO                          :    0 nagement short buffer num: 32
     16064 INFO                      wifi:    0 short buffer num: 32
         0 INFO                          :    0 32
         0 INFO                          :    0 namic tx buffer num: 32
     16081 INFO                      wifi:    0 buffer num: 32
         0 INFO                          :    0 
         0 INFO                          :    0 atic rx buffer size: 1600
     16097 INFO                      wifi:    0 uffer size: 1600
         0 INFO                          :    0  32
         0 INFO                          :    0 atic rx buffer num: 16
     16113 INFO                      wifi:    0 uffer num: 16
         0 INFO                          :    0 32
         0 INFO                          :    0 namic rx buffer num: 32
     16229 INFO                      wifi:    0 buffer num: 32
         0 INFO                          :    0 
         0 INFO                          :    0 softAP (9c:9c:1f:e3:eb:49)
     16235 INFO                      wifi:    0 c:9c:1f:e3:eb:49)
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 ower save buffer number: 16
     16252 INFO                      wifi:    0  buffer number: 16
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 x length of beacon: 752/752
     16270 INFO                      wifi:    0 of beacon: 752/752
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 x length of beacon: 752/752
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_user NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: web_login_pass NOT_FOUND
     16315 INFO               APState.cpp:   93 Setup access point.
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: ap_ssid NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: ap_passphrase NOT_FOUND
     17081 INFO                      wifi:    0 of beacon: 752/752
         0 INFO                          :    0 r num: 32
         0 INFO                          :    0 ower save buffer number: 16
     17097 INFO               APState.cpp:  186 SSID: pixelix
     17104 INFO            DisplayMgr.cpp:  997 Slot 0 (SysMsgPlugin) now active.
     27820 INFO                      wifi:    0  buffer number: 16
         0 INFO                          :    0 
         0 INFO                          :    0 0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
     27829 INFO                      wifi:    0 1,1>, ap:<1,1>, sta:<255,255>, prof:1
         0 INFO                          :    0 :1
         0 INFO                          :    0 : f0:18:98:37:b0:34 join, AID=1, bgn, 20
    153632 INFO                      wifi:    0 :37:b0:34 join, AID=1, bgn, 20
         0 INFO                          :    0 
         0 INFO                          :    0 : f0:18:98:37:b0:34 leave, AID = 1, bss_flags is 134243, bss:0x3ffdaf80
    153643 INFO                      wifi:    0 :37:b0:34 leave, AID = 1, bss_flags is 134243, bss:0x3ffdaf80
         0 INFO                          :    0 34243, bss:0x3ffdaf80
         0 INFO                          :    0 0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
    157178 INFO                      wifi:    0 ,0>, ap:<1,1>, sta:<255,255>, prof:1
         0 INFO                          :    0 34243, bss:0x3ffdaf80
         0 INFO                          :    0 0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
    157190 INFO                      wifi:    0 ,0>, ap:<1,1>, sta:<255,255>, prof:1
         0 INFO                          :    0 :1`

0ooYoo0, 2022-01-14:
Have you also flashed the filesystem image, just to be sure?

kuifje123, 2022-01-14:
No I didn’t. Where can I find this image?

0ooYoo0, 2022-01-14:
You have to generate and upload them as well.
Within the Project Tasks in VS Code e.g. 
- az-delivery-devkit-v4-usb-->Platform-->Build Filesystem Image
- az-delivery-devkit-v4-usb-->Platform-->Upload Filesystem Image

kuifje123, 2022-01-14:
Thanks, that did it!

closed 2022-01-14 by kuifje123