BlueAndi/Pixelix #70: Precompiled binarys for Wemos mini and ESP8266

Rushmed, 2022-02-04:
Hi.

Thanks for this project.
Is there a chance to offer precompiled .bin for ESP8266 and Wemos mini.

BlueAndi, 2022-02-05:
Hi,
you are welcome. Supporting the esp8266 is currently not on the roadmap.
The esp8266 sw platform is community driven and the esp32 is driven by espressif. Unfortunately this causes differences, e.g. the preferences component only exists on esp32 side. All sw and hw differences must be analyzed to create a hardware abstraction layer concept. Unfortunately nothing which can be done in a short time in my opinion.

Conclusion for the moment: No, I can not promise.

Rushmed, 2022-02-05:
Okay thaks!

closed 2022-02-05 by BlueAndi

Rushmed, 2022-02-09:
Is there a prcompiled ESP32 version?

BlueAndi, 2022-02-09:
No, not right now. We can open an issue and will plan it for the future. But there will be not only one esp32 binary, because we have to consider every esp32 variant and module, which means for consistency we have to provide several binaries.
And of course the binaries contain a fix LED topology.

BlueAndi, 2022-02-11:
See #72