BlueAndi/Pixelix #79: No Slots in Display

wendtc, 2022-02-27:
Hi there ,
What can you do with the Display menu in the web interface?
I don't see any slots
Christia
[image]
n

BlueAndi, 2022-02-27:
It seems your browser blocks the websocket communication. What browser do you use?

wendtc, 2022-02-27:
Firefox

BlueAndi, 2022-02-27:
Can you press F12 in the Firefox and have a look at the console which kind of errors there are?
Which version of Pixelix do you use?

wendtc, 2022-02-27:
Version 5.0.1 
I have tested other Version its the same.
[image]

BlueAndi, 2022-02-27:
Can you show the "Konsole" tab?
I tested it with Edge, Chrome, Safari (IPhone) and Samsung Browser without any issue. Hmm...
Did you erase the whole flash before programming filesystem and firmware?

wendtc, 2022-02-27:
I have testet with 3 new Chips - the same.
I don't know if that's a problem - the versioning doesn't work.
[image]
where is the command to read sent?

BlueAndi, 2022-02-27:
```~SW_VERSION~``` is replaced by the firmware with the correct version. Can you show me the log from the esp32 board?
How did you flash everything?

wendtc, 2022-02-27:
Visual Studio Code
Download the source via zip.

BlueAndi, 2022-02-27:
The log shows that the ```<ESP Async WebServer> 1.2.3``` is used, but that is invalid. This lead to the problem that all keywords surrounded with the tilde are not replaced and your browser throws the exception.

Did you change something in the platformio.ini?

BlueAndi, 2022-02-27:
[image]

BlueAndi, 2022-02-27:
Can you please upgrade your platformio? Because I see in your log other warning messages, which should not be there.
[image]

wendtc, 2022-02-27:
Ich hab in der platform ini zeile 67 +68 auskommentiert und die 2 libs in das lib verzeichnis kopiert wegen der meldungen unten .

Ich habe jetzt noch mal eine blanke installation genommen.
Git gab ich auch noch mal installiert.

[image]

BlueAndi, 2022-02-27:
Das Problem scheint platformio zu sein. Probiere folgendes:
1. Schliesse VSCode.
2. Lösche .platformio Verzeichnis in c:/Users/Christian bitte vollständig.
3. Öffne das Projekt in VSCode
4. Nun läuft die autom. Neuinstallation, bitte so lange warten, bis in der Konsole um einen Neustart von VSCode gebeten wird.
5. VSCode schliessen und nochmal Projekt öffnen.
6. Nun nochmal bauen und es sollte gehen.

wendtc, 2022-02-27:
Ich teste und melde mich

wendtc, 2022-02-27:
Super hat geklappt
Danke für die super Hilfe.
Daran hab ich gar nicht gedacht......
Christian

BlueAndi, 2022-02-27:
Alles klar, gerne.

closed 2022-02-27 by BlueAndi