BlueAndi/Pixelix #81: 8 MB partition table on TTGO

Giuli0ne, 2022-03-20:
Hi,
I'm testing on a TTGO with 4MB of Flash. I've found the default_8Mb.csv as the partitions for the TTGO - USB environment.
At the moment, neither this, nor the custom partition table is working.
Here's some numbers.
Application upload:
Wrote 1379680 bytes (784087 compressed) at 0x00010000

I tried with these partitions to optimize for the application size:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   , 1350K,
app1,     app,  ota_1,   , 1350K,
spiffs,   data, spiffs,  , 1390K,

But the File system says:
A fatal error occurred: File .pio\build\ttgo-t-display-usb\spiffs.bin (length 1423360) at offset 2949120 will not fit in 4194304 bytes of flash.

Thank you

BlueAndi, 2022-03-21:
The TTGO T-Display with 8 MB is a special one and Pixelix runs on its display. Which TTGO display do you have exactly? You may need to adapt e.g. the DevKit one by changing just the board type.

BlueAndi, 2022-03-27:
@Giuli0ne Any success?

Giuli0ne, 2022-03-27:
I could test only on an az-delivery esp32 with an 8x32 led matrix and the installation/setup was flawless.
I can spend some time on such projects only during the weekend, so sorry for the late reply.
Feel free to close the issue if it goes stale, I could open a more specific one (or revive this one) when I have more informations.
Thank you again

BlueAndi, 2022-03-28:
> I can spend some time on such projects only during the weekend, so sorry for the late reply.

No problem, I was just curious about the result. 😃 

> Feel free to close the issue if it goes stale, I could open a more specific one (or revive this one) when I have more informations.

Ok.

closed 2022-06-06 by BlueAndi