BlueAndi/Pixelix #82: automatic AP when wifi station is unavailable.

gzawahra, 2022-03-27:
hello ! pixelix is working great, thank you for all the hard work !
i'd like to suggest a simple improvement where in case it's impossible to connect to configured wifi station after n number of retries that that the display would automatically switch to AP mode, and then if no one connects to AP for x number of minutes to retry automatically to connect to configured SSID again, also wouldn't it be simpler if on first startup to automatically go to AP mode if no wifi is configured ? there seems to be a problem for me where when using an external touch button i can't seem to make go into AP mode since it takes longer to start up than a normal button so it wouldn't be low at esp boot. thanks for your consideration !

BlueAndi, 2022-03-27:
Nice to hear that you like the project. Thanks.

Its a security feature not to automatically start up in AP mode and must be forced manually via the user button. Usually configuring the network in AP mode should be a one time task for most of the users.

The remote user button can not be used to trigger AP mode at startup, because it requires a existing network connection to Pixelix, which does not exist at this time. :wink:

[image]

gzawahra, 2022-03-28:
thanks for the quick answer ! security concerns are completly valid in this case i haven't thought about it that way ! thanks !

closed 2022-03-28 by gzawahra