BlueAndi/Pixelix #86: 8 pixel font?

GRIETZEN, 2022-07-04:
Would it be possible to replace the current (5-pixel height) font to an 8-pixel font to improve readability?

file:///run/user/1000/gvfs/mtp:host=motorola_moto_g_9__play_ZY32BTLBJB/Almacenamiento%20interno%20compartido/DCIM/Camera/IMG_20220704_120947165_HDR.jpg

BlueAndi, 2022-07-04:
In principle it is possible, but several plugins would need adaptions. E.g. the date/time can not show the day of the week below date/time info. IconTextLampPlugin won't work, the same for the VolumioPlugin. Which font do you have in mind?

GRIETZEN, 2022-07-05:
I did a quick check on the source code of Pixelix but could no find any hint how and where the font is specified.
Just to give you an idea, something like this: 

https://github.com/TheRobotFactory/EightBit-Atari-Fonts

BlueAndi, 2022-07-05:
Search for "font" and you should find all relevant places. In general the TextWidget is used by the plugins and it uses TomThumb font as default:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/0eec32d71996096318bc8e529bda7f5f41612146/lib/YAWidgets/TextWidget.h#L91

https://github.com/BlueAndi/esp-rgb-led-matrix/blob/0eec32d71996096318bc8e529bda7f5f41612146/lib/YAWidgets/TextWidget.cpp#L65-L66

It provides a interface to change the font, which can be done by each plugin:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/0eec32d71996096318bc8e529bda7f5f41612146/lib/YAWidgets/TextWidget.h#L239-L250

Find the TomThumb font here: https://github.com/BlueAndi/esp-rgb-led-matrix/blob/master/lib/YAWidgets/TomThumb.h

If you like to try it, you need to convert the font into the right format. Years ago I converted it from the original again and updated it in the Adafruit gfx lib too: https://github.com/adafruit/Adafruit-GFX-Library/pull/268

You will find there the script I was using.

GRIETZEN, 2022-07-07:
Thanks, I will have a look into it....

BlueAndi, 2022-07-19:
@GRIETZEN I added the possibility to choose the font type ("default", "normal" or "large") per plugin instance.

Example:
```json
    {
      "name": "JustTextPlugin",
      "uid": 32690,
      "alias": "",
      "fontType": "large"
    }
```
Result:
[image]

See ./lib/Fonts.cpp the font definitions itself. For the 8pt font I used https://github.com/muwerk/mufonts

You can test it in the "Development" branch.

closed 2022-07-22 by BlueAndi