BlueAndi/Pixelix #89: SPIFFS Mount Failed -10025

YvesH87, 2022-10-19:
Hi,

after a lot of Troubleshooting i need to ask.
Is there an known Problem with SPIFFS`

I get the Message after Flashing

`        31 INFO             InitState.cpp:  415 PIXELIX starts up ...
        32 INFO             InitState.cpp:  416 SW version: v5.0.3:lc
        33 INFO             InitState.cpp:  417 SW revision: c0bae8e9375ec54c6ab148b037bdde6e7a96b5ae
        42 INFO             InitState.cpp:  418 ESP32 chip rev.: 1
        47 INFO             InitState.cpp:  419 ESP32 SDK version: v3.3.5-1-g85c43024c
        55 INFO             InitState.cpp:  420 Wifi MAC: 78:E3:6D:11:8A:60
        61 INFO             InitState.cpp:  421 LwIP version: 2.0.3
        69 ERROR                   SPIFFS:    0 mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
        79 FATAL            InitState.cpp:  144 Couldn't mount the filesystem.
        86 INFO            ErrorState.cpp:   73 Going in error state.
        92 INFO            ErrorState.cpp:   80 Low level error 4.`

Did i miss something or do i got Bad Luck with the ESP-Modules?

Change the hardware and Tried Erasing the Flash with 2 different Programs.
Same outcome...

BlueAndi, 2022-10-19:
Did you flash the filesystem first?

YvesH87, 2022-10-19:
Thats a good point. Thanks for the clarification.

closed 2022-10-19 by YvesH87