BlueAndi/Pixelix #97: Add more slots

Nikey, 2022-11-07:
It is now limited to 11 slots, is it possible to add more?
Why you hard limited the slots anyway?

BlueAndi, 2022-11-07:
It may run out of heap memory if you have more slots. Its still an embedded system with limited RAM. The usage of PSRAM like in the Adafruit Feather V2 may be the solution here. Currently PSRAM is not considered. Don't forget that you need enough heap space for the network communication as well.

If you have a fix plugin setup, you can increase the number of slots in the settings (see max. values there) and test it.

BlueAndi, 2022-11-07:
Just checked the sdkconfig and CONFIG_SPIRAM_USE_MALLOC is set. That means PSRAM is automatically considered. But need to verify that. ;-)

Nikey, 2022-11-07:
Okay totally understandable. I use all 10 slots atm and my free heap size on the debug page says 50%. So I think there should be more heap space left for more slots.
PSRAM could be another factor to raise the limit more, good idea :)

BlueAndi, 2022-11-08:
In your plugin setup there is more available, but combining it with others will result in lower heap space. :-)

Will think about how to consider the heap space to determine the limitation. With some modifications in the sources its possible to get some more free. I have some refactorings in mind, but I guess I will need some time to update it.

Nikey, 2022-11-08:
Plugin setup, what do you mean?

BTW: Thanks for this wonderful project, I really appreciate your work here. Would love to see this project growing and evolving - Is there any chance to reach you out on discord/skype? Want to help you with some tasks here!

BlueAndi, 2022-11-08:
With plugin setup I meant which plugin(s) you use right now.

Yes I have a discord account (BlueAndi), but seldom used. No dedicated Pixelix server yet. You are welcome and good to hear that you like the Pixelix project. :-) Support is always appreciated.

Nikey, 2022-11-08:
Okay good to hear!
A dedicated pixelix server on discord would be nice. Would appreciate if you add me: Painkiller#5852
can't add you without the number :)

BTT: I could try to determinate the plugin which use the most heap size ATM so we have the highest value of heap size used per slot. That could help to calculate a value for the max slots supported.

BlueAndi, 2022-12-29:
@Nikey Please test, see Development branch.

closed 2023-01-03 by BlueAndi