BlueAndi/Pixelix #102: Direct display of important messages

Nikey, 2022-11-13:
It would be cool if some important messages can be displayed instantly after a REST command is received.

ogski, 2022-11-24:
I would be interested in this feature as well. Ideally a 'slot' that would override all other slots until another REST command was received to revert to the original slots

BlueAndi, 2022-11-30:
Idea:
* Activate slot via ```POST /display/slot/<id>```
    * The slot will be only activated, if it contains a enabled plugin.
    * If a different slot is marked sticky, the requested slot will not be activated.
* Set slot sticky, which means the slot will be activated and slot scheduling is disabled. ```POST /display/slot/<id> sticky=true```
    * Only one slot can be marked sticky.
    * If plugin is removed from slot, sticky flag will be removed and slot scheduling resumes with next slot.
    * If a slot shall be set sticky and a different slot is already marked sticky, the sticky flag is moved to the requested one.
* Remove sticky flag and resume slot scheduling with next slot.: ```POST /display/slot/<id> sticky=false```

BlueAndi, 2022-12-04:
Implemented in Development branch. Will be available in christmas release.

closed 2022-12-04 by BlueAndi