BlueAndi/Pixelix #104: can't configure matrix type in webGUI

dikkedimi, 2022-11-14:
While I have the code running on an ESP32 Devkit, the CJMCU-8x8 matrices appear to only count pixels correctly on one axis. So it appears these matrices are serpentine, instead of row-by-row.

Unfortunately I am not able to modify this in the web GUI. Would be a nice feature.

dikkedimi, 2022-11-14:
Thanks for the response, I was happy to find the option in the code after some digging. Luckily the ESP remembers the wi-fi credentials after flashing, else it would be quite of a hassle to modify this setting. I guess it's not as critical since this should only have to be figured out once.

I guess I completely glossed over the answer in the FAQ. Maybe this info is in the 'wrong' section? Seems to me like this is info one would love to have when doing the initial setup. I'm used to have a config.h or main.h file containing all these settings (but maybe I'm spoiled).

for anyone else ending up here:

`NeoTopology<RowMajor270Layout>                                             m_topo; // for CJMCU-8*8`

made it display something other than nonsense :)

BlueAndi, 2022-11-14:
The matrix topology can not simply be setup during runtime, which depends on the underlying NeoPixelBus library. All kind of configurations must be present at compile time, which needs a lot of space. On the other hand the NeoPixelBus handles it very efficient, thats why it was choosen.

It seems most of the users have the same matrix topology as provided. :-)

closed 2022-11-23 by BlueAndi