BlueAndi/Pixelix #117: button not recognized

xcjantv, 2023-01-26:
I connected everything, but the pushbutton isn`t recognized.
The LEDs are flashing white here and there, but the button input isn`t recognized.
PLS help

BlueAndi, 2023-01-26:
Which development board?
Which SW version?
Where is the button connected to?
Is it connected to ground on the other side?
If you keep the button pressed about at least 1 min. from startup on, do you see the Pixelix wifi network?

Detail please ;-)

xcjantv, 2023-01-26:
> Which development board? Which SW version? Where is the button connected to? Is it connected to ground on the other side? If you keep the button pressed about at least 1 min. from startup on, do you see the Pixelix wifi network?
> 
> Detail please ;-)

ESP32 DEV module
latest version, I downloaded it yesterday
button connected to g4 and ground
nope I dont see the wifi network

BlueAndi, 2023-01-26:
Did you flash the firmware AND the filesystem too?

xcjantv, 2023-01-26:
Yes both

BlueAndi, 2023-01-26:
If you see only some white pixels, the display orientation is wrong. It needs to be adapted. If button doesn't react, there may be a different problem. What does the serial log shows?

xcjantv, 2023-01-27:
`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
        17 INFO             InitState.cpp:  446 PIXELIX starts up ...
        18 INFO             InitState.cpp:  447 Target: esp32doit-devkit-v1-usb
        28 INFO             InitState.cpp:  448 SW version: v6.0.0:lc
        28 INFO             InitState.cpp:  449 SW revision: 35b6e9ae84c30a0d60b4636cb99e6e12a30f340f
        39 INFO             InitState.cpp:  450 ESP32 chip rev.: 1
        49 INFO             InitState.cpp:  451 ESP32 SDK version: v4.4.2
        50 INFO             InitState.cpp:  452 Wifi MAC: 9C:9C:1F:C6:57:D4
        60 INFO             InitState.cpp:  453 LwIP version: 2.1.2esp

xcjantv, 2023-01-27:
I uploaded a [video](https://vimeo.com/793362382) of the white flickering

xcjantv, 2023-01-27:
I`m building [this](https://www.instructables.com/Lazy-Mini-Grid/) matrix, which has a "special" led order

BlueAndi, 2023-01-27:
I think the SW is up and running. It seems for me that the button may be connected to the wrong pin (I assume you pressed it long enough so the pin gets a low level detected).

Regarding the white flickering, its because the LED topology is wrong and I guess the number of LEDs are wrong. Pixelix out of the box expects a 8x32 matrix. You need to adapt it in the code to your needs.

Changing topoloy: https://github.com/BlueAndi/esp-rgb-led-matrix#where-to-change-panel-topology-of-the-led-matrix
Changing number of LEDs on x- and y-axis: ./esp-rgb-led-matrix_findBug/lib/HalLedMatrix/src/Board.h

BlueAndi, 2023-01-30:
One more ... can you keep the button pressed, performing a reset and checking the serial output? It should not show entering the idle state.

BlueAndi, 2023-02-02:
Any success?

closed 2023-02-06 by BlueAndi