BlueAndi/Pixelix #119: [Feature]

ddijanic, 2023-02-11:
add some basics,
in Just Text Plugin, to set text color, maybe color efect, fade colors
ADD EFFECT SCROLL plugin
DateTimePlugin add colorful date, hour, day

simple things and a lot is gained

BlueAndi, 2023-02-13:
Some basics are already there, e.g. https://github.com/BlueAndi/esp-rgb-led-matrix#how-to-change-text-properties

Will think about the possibility to have more stuff configurable. A general concept is necessary to avoid its different for every plugin type and it must consider multiple instances of the same plugin type.

For the effect plugin, a PR is welcome.

BlueAndi, 2023-02-27:
@ddijanic Please test latest commit in Development branch. DateTimePlugin can be colorized by using the text color properties in the date and time format strings. Additonal the colors for the day of the week can be set too. See in the plugin configuration JSON file and modify via file editor in the web interface.

ddijanic, 2023-03-09:
COOL, can you give me an example of just how edit in configuration json should look.

Is it normal to lose the text in the just text plugin after a power reset?
the plug-in is empty, and you have to type in the text again.

BlueAndi, 2023-03-10:
> COOL, can you give me an example of just how edit in configuration json should look.

```json
{
  "cfg": 0,
  "timeFormat": "\\#ff0000%H:\\#0000ff%M",
  "dateFormat": "%d.%m.",
  "dayOnColor": "#D3D3D3",
  "dayOffColor": "#404040"
}
```

In the timeFormat you can use the standard possibility to add colors, see https://github.com/BlueAndi/esp-rgb-led-matrix#how-to-change-text-properties
and how the time shall look like, see https://github.com/BlueAndi/esp-rgb-led-matrix/tree/Development#how-to-configure-the-datetime-format

> Is it normal to lose the text in the just text plugin after a power reset?
> the plug-in is empty, and you have to type in the text again.

This is intended behaviour, because usually in the background the information is periodically updated. Additional it prevents a high number of flash writes in case someone updates it very quickly 24/7/365.

Note, its currently only in the Development branch, therefore you need to switch the branch to see the updated FAQ or the plugin description for DateTimePlugin.

BlueAndi, 2023-03-10:
@ddijanic And after reading your post again I see you mentioned the configuration.json :-), please open the <plugin-uid>.json ;-)
You will find the plugin-uid in the display webpage.

ddijanic, 2023-04-02:
works, you cant close it. thanks.

when you drag the items over
GET ip/rest/api/v1/display/uid/<PLUGIN-UID>/dateTime
does not show this data
"timeFormat": "\\#ff0000%H:\\#0000ff%M",
   "dateFormat": "%d.%m.",
   "dayOnColor": "#D3D3D3",
   "dayOffColor": "#404040"
. show the basic data as in the master version, just show
{
   "data": {
     "cfg": 0
   },
   "status": "ok"
}
ther is on timeFormat,dateFormat,dayOnColor,dayOffColor

it would be great if we could in DateTimePlugin.html
set
timeFormat
dateFormat
dayOnColor
dayOffColor
like OpenWeatherPlugin.html

BlueAndi, 2023-04-11:
093417f04539cc36b199acdaf0f542184690f12f

closed 2023-04-11 by BlueAndi