BlueAndi/Pixelix #122: build for  Lolin S2 Mini ESP32-S2[Issue]

ddijanic, 2023-03-24:
I bought esp32 with 2 mini thinkers, that it will work.
but s2 is a bit different when it comes to sending programs to it.

I struggled for 2 days, I didn't succeed much.
if someone knows how to set this up, if not, wait again +30 days
but I managed to enter wled, they have it for s2 build.

this is how it looks with wled

[esp32s2]
build_flags = -g
  -DARDUINO_ARCH_ESP32
  -DARDUINO_ARCH_ESP32S2
  -DCONFIG_IDF_TARGET_ESP32S2
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -DCO
  -DARDUINO_USB_MODE=0 ;; this flag is mandatory for ESP32-S2 !
  ;; please make sure that the following flags are properly set (to 0 or 1) by your board.json, or included in your custom platformio_override.ini entry:
  ;; ARDUINO_USB_CDC_ON_BOOT, ARDUINO_USB_MSC_ON_BOOT, ARDUINO_USB_DFU_ON_BOOT

[env:lolin_s2_mini]
platform = espressif32@5.1.1
board = lolin_s2_mini
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} ${esp32s2.build_flags} #-D WLED_RELEASE_NAME=LolinS2
  -DBOARD_HAS_PSRAM
  -D ARDUINO_USB_CDC_ON_BOOT
  -D WLED_USE_PSRAM
  -D WLED_WATCHDOG_TIMEOUT=0
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -D LEDPIN=16
  -D BTNPIN=18
  -D RLYPIN=9
  -D IRPIN=7
  -D HW_PIN_SCL=35
  -D HW_PIN_SDA=33
  -D HW_PIN_CLOCKSPI=7
  -D HW_PIN_DATASPI=11
  -D HW_PIN_MISOSPI=9
;  -D STATUSLED=15
lib_deps = ${esp32s2.lib_deps}

BlueAndi, 2023-03-24:
I don't have a Lolin S2 Mini, but this is how I would configure it for the first run and please note, I didn't checked for the pinning!

```
; ********************************************************************************
; WEMOS Lolin S2 Mini - LED matrix
; ********************************************************************************
[board:wemos_lolin_s2_mini]
extends = common:esp32_env, display:led_matrix, config:small
board = lolin_s2_mini
board_build.partitions = ./partitionTables/custom_4MB.csv
board_build.filesystem = littlefs
build_flags =
    ${common:esp32_env.build_flags}
    ${display:led_matrix.build_flags}
    -DARDUINO_USB_MODE=0
    ${config:small.build_flags}
lib_deps =
    ${common:esp32_env.lib_deps_builtin}
    ${common:esp32_env.lib_deps_external}
    ${display:led_matrix.lib_deps_builtin}
    ${display:led_matrix.lib_deps_external}
    ${config:small.lib_deps}
lib_ignore =
    ${common:esp32_env.lib_ignore_builtin}
    ${common:esp32_env.lib_ignore_external}
    ${display:led_matrix.lib_ignore_builtin}
    ${display:led_matrix.lib_ignore_external}
extra_scripts =
    ${config:small.extra_scripts}
    pre:./scripts/get_git_rev.py

; ********************************************************************************
; WEMOS Lolin S2 Mini - Programming via USB
; ********************************************************************************
[env:wemos_lolin_s2_mini]
extends = board:wemos_lolin_s2_mini, common:prog_usb, mode:selected
build_flags =
    ${board:wemos_lolin_s2_mini.build_flags}
    ${mode:selected.build_flags}
extra_scripts =
    ${board:wemos_lolin_s2_mini.extra_scripts}
    ${common:prog_usb.extra_scripts}
```

BlueAndi, 2023-03-25:
Ahh ... and I realized in this moment, the esp32-s2 is only a single core mcu. :-/

ddijanic, 2023-03-27:
thank you for your effort and time.
he translated it, i sent it to esp.
I did not see the wifi host, after holding buten on pin 4.
I'll try it today with the panel to see if it works.|
serial monitoring does not work on esp s 2, of course :D
and I use the Development branch,
I will let you know what happens...

ps.
I tried the SHT3x sensor, it says it's 30 degrees, but it's 22 :D
then I found one DHT 11 from the other, he writes 23-24.
temperature offset would be good maybe...

BlueAndi, 2023-03-27:
I pushed the WEMOS Lolin S2 Mini support to Development branch.
Serial connection should work after a power reset. The USBCDC driver will be initialized.

Did you test the sensors with the Lolin S2 or another board? I am confused ... thought the Lolin S2 is not working right now.

Did you checked the datasheet of the SHT3x regarding accuracy, etc? I will have a look whether I have a SHT3x somewhere here to verify it too.

ddijanic, 2023-03-28:
Sensors tested whit ESP32-WROOM-32U, whit build esp32doit-devkit-v1.
temperatura works but dht11 is off by 2 degrees. SHT3x is off by 8 degrees..

Datasheet SHT3x-DIS
SHT30 Accuracy tolerance1 typ., 0°C to 65°C 0.3 °C
SHT31 Accuracy tolerance1 typ., -40°C to 90°C 0.3 °C
SHT35 Accuracy tolerance1 typ., -40°C to 90°C ±0.2 °C
Repeatability2 --------------- Low 0.24 °C
--------------------------------Medium 0.12 °C
------------------------------- High 0.06 °C
Resolution-------Typ -------- 0.015 °C

ddijanic, 2023-03-29:
WEMOS Lolin S2 Mini  Development branch
not working, upload 100% and reboot , but no error on panel E4.
after upload littlefs files, not working.
on board led fleshing, but no output on led panel.
serial monitor not working for this bord, so canot se whats the problem.
reboot with pin 4 to gnd. no wifi hotspot.
thanks we tried....

PS. 
I tried yesterday. 
to replace one image from OpenWeatherPlugin. 
WITH fire.bmp AND ADDED fire.sprite. 
I wanted to see if the animation could go along,
To make clouds that move etc for the openweatherplugin.
But fire was there, but it wasn't animation of course

I didn't find where it says how to insert animation into IconTextPlugin or somewhere.
it only says how to make it, not how to put it :D

BlueAndi, 2023-03-29:
> not working, upload 100% and reboot , but no error on panel E4.

Does that mean you saw the E4 on the panel or no error at all?

> on board led fleshing, but no output on led panel.

The Pixelix was in the error state. I updated the Development branch. Try again, the onboard LED will show the error number in case its an low level error.

> reboot with pin 4 to gnd. no wifi hotspot.

[image]

Hmm ... maybe the button driver failed. You will see it with the error id flashed with the onboard LED.

---

Where did you connect the LED matrix? The WEMOS Lolin S2 has no GPIO27, as I can see. Did you update the ./lib/HalLedMatrix/src/Board.h ?

> thanks we tried....

Means you give up? :-)

> I tried yesterday.
> to replace one image from OpenWeatherPlugin.
> WITH fire.bmp AND ADDED fire.sprite.
> I wanted to see if the animation could go along,
> To make clouds that move etc for the openweatherplugin.
> But fire was there, but it wasn't animation of course

The OpenWeatherPlugin doesn't support the spritesheet at the moment. I thought about it as well, but was not priority 1.
Currently I am facing the problem that for 4 MB flash modules, the flash is nearly full. But of course with 8 MB or more, the plugin can be updated to support it.

> I didn't find where it says how to insert animation into IconTextPlugin or somewhere.
> it only says how to make it, not how to put it :D

https://github.com/BlueAndi/esp-rgb-led-matrix#how-can-i-use-animated-icons
Only short I know ... :-)

> this is very cool and eazy https://www.piskelapp.com/p/create/sprite
> it has png,gif,
> and .C file out put, maybe it would perhaps be easiest to add C think :). for OpenWeatherPlugin, IconTextPlugin and new plug 8x 32 pic and animation plug :D

I know the PISKEL editor. The drawback with png and gif is the much more complex handling of the formats. The bitmap file format is easier and don't need to much power and space. Providing them as .c files  makes it inflexible, because you need to compile it in everytime you change. Please consider that there are not so many libraries somewhere providing the features for an MCU, like the esp32. And please consider the license model of them as well. Goal is to stay with MIT or compatible license. The bitmap parsing I implemented by my own, to get it with low space consumption and fast loading.

IconTextPlugin and IconTextLampPlugin e.g. support .bmp and spritesheets. The icons can be easily updated via webinterface, because they are in the filesystem.

ddijanic, 2023-03-30:
> Means you give up? :-)>
### no. 

Uploading .pio\build\wemos_lolin_s2_mini\firmware.bin
Wrote 1318288 bytes (829671 compressed) at 0x00010000 in 35.3 seconds (effective 298.7 kbit/s)...
Hash of data verified.

Uploading .pio\build\wemos_lolin_s2_mini\littlefs.bin
Wrote 1310720 bytes (316833 compressed) at 0x002b0000 in 15.7 seconds (effective 667.7 kbit/s)...
Hash of data verified.

https://ibb.co/hm6d4fR
only with build and after littlefs.bin, it shows the same
led pin 21

on bord led is blinking, long on, off short, on short, off short, on long.

wifi ap mod not working, 4 to gnd.

PISKEL editor white c output make text liki this for 1 frame of 8x8 grid:
0x00000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0xff000000, 
0x00000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0xff000000, 
0x00000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 0xff000000, 0xff000000, 
0x00000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 
0x00000000, 0x00000000, 0x00000000, 0xff000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 
0x00000000, 0x00000000, 0xff000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 
0x00000000, 0xff000000, 0xff000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000, 
0xff000000, 0x00000000, 0x00000000, 0x00000000, 0xff000000, 0x00000000, 0x00000000, 0x00000000

we can't use text only because of the license?

Ldr
I tried using ky-018, I get between 0.5v and 3v on pin 34.
and the sensor plug doesn't see the sensor, and I can't turn on the automatic light intensity.

I'm waiting for it to arrive from China LdrGl5528

BlueAndi, 2023-03-30:
> only with build and after littlefs.bin, it shows the same
> led pin 21 (I tried and pin 38 behaves the same)

Which LED matrix orientation does your matrix requires?

> on bord led is blinking, long on, off short, on short, off short, on long.

My fault, had in mind the LED is low active. Fixed in Development branch.
But it looks like the I2C pins are wrong and yes, they are. I updated the pinning to
* SDA 33
* SCL 35
* LDR 1
* LED matrix data out 21
* I2S removed

> wifi ap mod not working, 4 to gnd.

Yes, as long as the board is in error state, the wifi ap won't be spawned.

> PISKEL editor white c output make text liki this for 1 frame of 8x8 grid:

Not as efficient than as binary like bmp. :-)

> we can't use text only because of the license?

No, I meant the libraries for handling png, gif and etc. formats.

> I tried using ky-018, I get between 0.5v and 3v on pin 34.
> and the sensor plug doesn't see the sensor, and I can't turn on the automatic light intensity.

With the esp32doit-devkit-v1 board, I guess? We should not mix it here, to not confuse ourself. :-)

ddijanic, 2023-03-30:
> Which LED matrix orientation does your matrix requires?
matrix is okay, it works with master build.

With the esp32doit-devkit-v1 board, I guess? We should not mix it here, to not confuse ourself. :-)
yes, I use esp32doit-devkit-v1 board for test, so i know matrix is okey :)
but ldr not working, and dht11 temp offset  would be nice to :)

ddijanic, 2023-03-31:
https://ibb.co/FDkNW0h

BlueAndi, 2023-03-31:
Great, its working.

As its getting darker there is  a certain threshold after the LDR is not recognized anymore. Caused by low voltage and bad ADC accuracy. Therefore I am thinking about removing the automatic LDR detection and replace it with a manual activation in the settings.

The display brightness doesn't react immediately on brightness changes by intention. There are two kind of filters. One detects short term changes and the other long term changes. If the tendence of both goes into the same direction, the brightness will be adjusted smooth. 
See
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/Development/src/Gfx/BrightnessCtrl.h
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/Development/src/Gfx/BrightnessCtrl.cpp

> can you give me the code just how to turn on spritesheets in OpenWeatherPlugin?

You will find a simle example in the IconTextPlugin ... see e.g. this part:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/e8734273196b778a55e127247a15f6f71fb122ac/lib/IconTextPlugin/src/IconTextPlugin.cpp#L189-L196

I updated the Development branch again and refactored a little bit for easier maintenance. Please check again.
If its successful, I would like to close this issue regarding the S2 mini board.

You can use the discord channel for general discussion: https://discord.gg/Xe5bGnQJ
Note,  sometimes I need some time to answer, as there is still family life around. ;-)

ddijanic, 2023-03-31:
OpenWeatherPlugin is broken on dev s2 , its active is set up. 

  "apiKey": "87e6a2b3ddd528e530acfd1cfc3b68e3",
  "lat": "46.100376 ",
  "lon": "19.833549",
  "other": 0,
  "units": "metric"

but its showing onley ?

btc plug works. sunsetworks, all works except **OpenWeatherPlugin**

ddijanic, 2023-03-31:
> As its getting darker there is a certain threshold after the LDR is not recognized anymore. Caused by low voltage and bad ADC accuracy. Therefore I am thinking about removing the automatic LDR [...]
this work great.
https://github.com/Aircoookie/WLED/tree/main/usermods/SN_Photoresistor
you have little to play around with the whales, but after 3 minutes its done
I have made 2 clock whit this and its good.

I've seen a modified one that's even better, I think it's easier to adjust. I need to find where that one is.

ddijanic, 2023-03-31:
> 
> I updated the Development branch again and refactored a little bit for easier maintenance. Please check again. If its successful, I would like to close this issue regarding the S2 mini board.
> 

build works,  OpenWeatherPlugin not.....

> You can use the discord channel for general discussion: https://discord.gg/Xe5bGnQJ Note, sometimes I need some time to answer, as there is still family life around. ;-)

DEAL :)

BlueAndi, 2023-03-31:
Thanks for the feedback.

closed 2023-03-31 by BlueAndi

ddijanic, 2023-04-02:
for spritesheets in OpenWeatherPlugin

if (false == m_bitmapWidget.loadSpriteSheet(FILESYSTEM, getFileName(FILE_EXT_SPRITE_SHEET), getFileName(FILE_EXT_BITMAP))) 
     { 
         (void)m_bitmapWidget.load(FILESYSTEM, getFileName(FILE_EXT_BITMAP)); 
     } 
add this at top
/* Initialize sprite sheet parameter filename extension. */
const char* IconTextPlugin::FILE_EXT_SPRITE_SHEET   = ".sprite";

not complain build..

BlueAndi, 2023-04-04:
@ddijanic I asked you already on Discord about it. :-)