BlueAndi/Pixelix #140: [Feature]

lilxanman, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I am hoping for a generic plugin that displays information given from a api request that is provided by the user. I would like the ability to use display api information like YouTube subscribers, Weather Hazzards, etc.

**Describe alternatives you've considered**
Maybe instead of API, use RSS feeds

**Additional context**
Add any other context or screenshots about the feature request here.

BlueAndi, 2023-07-26:
In the Development branch you will find two kind of plugins, providing this:
* GrabViaRestPlugin
* GrabViaMqttPlugin

Please check them out ... the documentation may not be up to date. Any improvement is welcome. ;-)

BlueAndi, 2023-07-26:
And consider the other way around as well, by using e.g. node-red, homeassistant and etc. to show the information on the display.

lilxanman, 2023-07-26:
Am I meant to change the code, or should it just appear like a regular plugin? Im fine with both im just unsure

lilxanman, 2023-07-26:
So i've found my answer. I feel it is worth noting that names of the plugins are configured to be the same

lilxanman, 2023-07-27:
Also do you have any info on the Json Filters

BlueAndi, 2023-07-28:
> So i've found my answer. I feel it is worth noting that names of the plugins are configured to be the same

They are slightly different ... one is for REST APIs and the other for MQTT.

> Also do you have any info on the Json Filters

Here is an example for github stargazers, I hope this helps.

```json
{
  "method": "GET",
  "url": "https://api.github.com/repos/BlueAndi/esp-rgb-led-matrix",
  "filter": {
    "stargazers_count": true
  },
  "iconPath": "/configuration/github.bmp",
  "format": "\\calign%s",
  "multiplier": 1,
  "offset": 0
}
```

The filter is used to determine where the information can be found, which shall be shown on the display.

BlueAndi, 2023-08-02:
Is it working for you?

BlueAndi, 2023-08-11:
@lilxanman ?

closed 2023-09-14 by BlueAndi