BlueAndi/Pixelix #142: [Bug] Countdown Plugin Restart Issues

lilxanman, 2023-07-30:
**Describe the bug**
A clear and concise description of what the bug is.

When you save the date on the countdown plugin, the correct information shows on the matrix, when the esp32 repowers, the countdown plugin will display a "?". I've found that when I go into the web panel, that the information I entered is there.

Note: When I click the update button, the date shows again until it reboots.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Plugins > CountdownPlugin
2. Enter in details
3. Click Update
4. Restart The Device

**Expected behavior**
The correct countdown information should show upon reboot

**Please complete the following information:**
 - Development board: esp32doit-devkit-v1
 - Version: v7.1.1:lc

**Additional context**
Add any other context about the problem here.

**Log**
If applicable, add serial log output to support the analysis.

BlueAndi, 2023-07-31:
I can not reproduce it with the official release v7.1.1, but in the Development branch. You wrote v7.1.1:lc, which means official release with local changes. Do you use the main or the development branch?

The "?" will stay in case the time was not synchronized by the NTP.

BlueAndi, 2023-08-02:
I fixed the problem in the Development branch. Please verify and if the problem still exists, reopen the issue.

closed 2023-08-02 by BlueAndi