BlueAndi/Pixelix #143: [Bug]Countdown Plugin

skud75, 2023-09-04:
The Countdown plugin after setting, every ESP32 restart, it shows a '?' instead the days number remain. It need to be set again. 

Steps to reproduce the behavior:
1. Go to display and install Countdown Plugin
2. Click on Plugin and Countdown and set a new target data
3. It show a corretct days remaining
4. Restart ESP32 (by reset or power off/on)
5. See error : After restart display shows : "?"
6. Set again Countdown plugin and display will be correct till new reset

*Expected behavior*
The display must show remaining days also after power on/off

*Please complete the following information:*
 - Development board: ESP32-WROOM-32
 - Version: v7.1.1:b
 - ESP SDK Version: v4.4.4
 - Flash Chip Size : 4 MByte

*Additional context*
Add any other context about the problem here.

skud75, 2023-09-04:
I also verify the configuration file stored in the file system (<uid>.json). It's correct

BlueAndi, 2023-09-05:
I made some bugfixes in the Development branch, can you verify with that again?

skud75, 2023-09-05:
How can i do that? i downloaded the lastone github version.

BlueAndi, 2023-09-05:
I compiled it for you (rev. 8c06094f7c8a87e514b294cd8ab65ecf4fa557d6)
[esp32doit-devkit-v1.zip](https://github.com/BlueAndi/esp-rgb-led-matrix/files/12527686/esp32doit-devkit-v1.zip)

skud75, 2023-09-06:
With Development branch Version works fine. Thanks a Lot.

BlueAndi, 2023-09-06:
Thanks for the feedback. I will need some more days to be able to release a new version, but hope its ok for you so far.

closed 2023-09-06 by BlueAndi