BlueAndi/Pixelix #146: [Bug] REST API: /display/slot/{id} always returns 'Invalid slot id.'

Nikey, 2023-09-30:
**Describe the bug**
POST with sticky=true to /display/slot/{id} always returns
`{
  "error": {
    "msg": "Invalid slot id."
  },
  "status": "error"
}`
even if the ID is correct

**To Reproduce**
POST to /display/slot/{id}

**Expected behavior**
Return code 200 with 
`{
  "data": {},
  "status": "ok"
}`

BlueAndi, 2023-09-30:
I tested with the development branch and I can set slots sticky with the expected behaviour:
```bash
$ curl -d "sticky=true" -X POST http://192.168.YYY.XXX/rest/api/v1/display/slot/2
{
  "data": {},
  "status": "ok"
}
```

Which version do you use?
Does the problem appears after restart in the first request or after some time?

Nikey, 2023-10-01:
okay i misunderstood the id param 🤦 
i thought i have to use the id from the plugin...
Sorry!

closed 2023-10-01 by Nikey