BlueAndi/Pixelix #148: [Feature] Add rotate display 180deg for T-Display-S3

P1X3L8, 2023-10-31:
I didn't see any easy way of doing this in the code. TY

nhjschulz, 2023-11-01:
Is it is only for the display on the T-Display-S3 and you don't need dynamic configuration through the UI ?  Then this single line change should do the trick:

Add the line "m_tft.setRotation(2);" into  file lib\HalTftDisplay\src\Display.h, function begin() as shown below.
```
/**
     * Initialize base driver for the display.
     *
     * @return If successful, returns true otherwise false.
     */
    bool begin() final
    {
        m_tft.init();
        m_tft.setRotation(2);
        m_tft.fillScreen(TFT_BLACK);

        return true;
    }
```

P1X3L8, 2023-11-01:
Hi nhjschulz, this worked, thank you!

BlueAndi, 2023-11-05:
Thanks nhjschulz for providing this feature.

closed 2023-11-05 by BlueAndi