BlueAndi/Pixelix #152: [Bug] MQTT looses connection

Nikey, 2023-12-03:
**Describe the bug**
MQTT loose connection after a few hours with mosquitto in Homeassistant. (tested in development branch)

**To Reproduce**
Steps to reproduce the behavior:
1. Add MQTT connection to Homeassistant
2. Grab sensor data
3. Sensor data is grabbed but looses the connection for some reason after some hours

**Expected behavior**
Sensor data is grabbed constantly.

**Screenshots**
[image]

**Additional context**
The MqttApiTopicHandler.cpp seems to send data after it looses connection (see log below), but HA don't recognize it for some reason.

A restart of mosquitto resolve the problem for another few hours. Unfortunatly don't have a log output from it..

**Log**
38534601 | INFO | MqttApiTopicHandler.cpp | 417 | Published: pixelix/sensors/1/humidity/state
38531861 | INFO | MqttApiTopicHandler.cpp | 417 | Published: pixelix/sensors/2/illuminance/state
38521776 | INFO | MqttApiTopicHandler.cpp | 417 | Published: pixelix/sensors/2/illuminance/state
38516634 | INFO | MqttApiTopicHandler.cpp | 417 | Published: pixelix/sensors/0/temperature/state
38511743 | INFO | MqttApiTopicHandler.cpp | 417 | Published: pixelix/sensors/2/illuminance/state

Nikey, 2023-12-09:
It unfortunatly didn't happen again. Did the update now, will reopen this issue with more information when the issue occured again.
Thanks for your fast fixes :)

BlueAndi, 2023-12-09:
> It unfortunatly didn't happen again. Did the update now, will reopen this issue with more information when the issue occured again. Thanks for your fast fixes :)

Ok, you are welcome.

Nikey, 2024-01-02:
It seems like it sometimes read false data from sensors ("nan" instead of the value of sensor, see logs below).
The error happened on version 7.2.0

> Logger: homeassistant.components.mqtt.models
> Source: components/mqtt/models.py:305
> Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
> First occurred: 1. Januar 2024 um 15:14:25 (3 occurrences)
> Last logged: 10:21:24
> 
>     Exception raised when updating state of sensor.sensors_0, topic: 'pixelix/sensors/0/temperature/state' with payload: b'{"value":"NAN"}'
>     Exception raised when updating state of sensor.sensors_1, topic: 'pixelix/sensors/1/humidity/state' with payload: b'{"value":"NAN"}'
> 
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 305, in process_write_state_requests
>     entity.async_write_ha_state()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 745, in async_write_ha_state
>     self._async_write_ha_state()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 845, in _async_write_ha_state
>     state, attr = self._async_generate_attributes()
>                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 786, in _async_generate_attributes
>     state = self._stringify_state(available)
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 751, in _stringify_state
>     if (state := self.state) is None:
>                  ^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 634, in state
>     raise ValueError(
> ValueError: Sensor sensor.sensors_0 has device class 'temperature', state class 'None' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'

BlueAndi, 2024-01-02:
@Nikey Can you open a separate issue for this, because its a different topic. Thx.

Nikey, 2024-01-06:
Saw you already created one, will close this again.

closed 2024-01-06 by Nikey