BlueAndi/Pixelix #164: [Feature] Add new Board   ESP32 S3 DevKitC 1 N16R8 WROOM

d0m1n1qu3, 2024-04-22:
Hi

i was searching for an ESP32 with more that 4MB flash. so i choose this one    ESP32 S3 DevKitC 1 N16R8 WROOM

https://www.amazon.de/dp/B0CLD4QKT1

but i can't get it run (blinky and so on works) .. 

i tried with the base from build config from   lilygo-t-display-s3     but i get only a reboot loop

any ideas ? 

thanks for helping :-) 

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_Sad:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c3
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a029
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a020
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021`

d0m1n1qu3, 2024-04-23:
i tried to build a board definition for this ESP board .. and also configs for pixelix ..

now i get this error .. my be i forgot some configuration, because of hostname not a string   etc..

`
ESP_ERROR�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

        18 INFO             InitState.cpp:00486 PIXELIX starts up ...
        18 INFO             InitState.cpp:00487 Target: esp32-s3-devkitc-1-N16R8
        28 INFO             InitState.cpp:00488 SW version: v7.2.2:lc
        58 INFO             InitState.cpp:00490 SW revision: 3780ad6ec88a1a9ec0a17204ad7553ce70e415c6
        58 INFO             InitState.cpp:00491 ESP32 chip rev.: 0
        68 INFO             InitState.cpp:00492 ESP32 SDK version: v4.4.5
        88 INFO             InitState.cpp:00494 Wifi MAC: 80:65:99:xx:xx:xx
        88 INFO             InitState.cpp:00495 LwIP version: 2.1.2esp
[   455][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ha_dp NOT_FOUND
       170 INFO    TopicHandlerService.cpp:00073 Topic handler service started.
       180 INFO       SettingsService.cpp:00328 Settings service started.
       180 INFO              AudioDrv.cpp:00143 Audio driver task is up.
       648 INFO              AudioDrv.cpp:00189 I2S driver installed.
       658 INFO          AudioService.cpp:00103 Audio service started.
[   487][E][Preferences.cpp:483] getString(): nvs_get_str len fail: mqtt_broker_url NOT_FOUND
[   492][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
       681 INFO           MqttService.cpp:00117 MQTT service started.
       691 INFO              ClockDrv.cpp:00089 No RTC is available.
[   521][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timezone NOT_FOUND
[   521][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ntp_server NOT_FOUND
[   528][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
[  5735][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  5736][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  5737][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (5923) gpio: gpio_set_level(226): GPIO output gpio_num error
[  5768][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
      6038 INFO    SensorDataProvider.cpp:00418 Sensor GL5528: available
      6038 INFO    SensorDataProvider.cpp:00418 Sensor SHT3x: -
      6048 INFO    SensorDataProvider.cpp:00418 Sensor DHT11: -
      6048 INFO    SensorDataProvider.cpp:00418 Sensor Battery: -
      6059 INFO    RestApiTopicHandler.cpp:00102 Register: /rest/api/v1/sensors/2/illuminance
      6059 INFO    MqttApiTopicHandler.cpp:00091 Register: pixelix-99c84dcc/sensors/2/illuminance
[  5890][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
      6089 INFO            DisplayMgr.cpp:00213 DisplayMgr is up.
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
     ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
[... cut ...]
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d6ff
file: ".pio/libdeps/esp32-s3-devkitc-1-N16R8/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
`

d0m1n1qu3, 2024-04-23:
i have played around an added a fork with my changes .. maybe some one can have a look an give me a hint :-)

https://github.com/d0m1n1qu3/esp-rgb-led-matrix/commit/0333e0129fd92943301f9b5c042809191880d0d6

BlueAndi, 2024-04-23:
I didn't check your change yet, but can you please just try this branch and select your board there?
The pinning might not be correct yet ...
https://github.com/BlueAndi/esp-rgb-led-matrix/tree/support/esp32-s3-devkitc-1-n16r8v

d0m1n1qu3, 2024-04-23:
hi. i changed the pin for LED to 14U and got it more working than my fork ..

hier the actual outputs ..

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f6d6a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

[   303][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
18         INFO    InitState.cpp:486            PIXELIX starts up ...
28         INFO    InitState.cpp:487            Target: esp32-s3-devkitc-1-n16r8v
28         INFO    InitState.cpp:488            SW version: v7.2.1:lc:b
58         INFO    InitState.cpp:490            SW revision: 89c9fc776aa36a8cb09fbdeb99fbf4fd49ff8d93
58         INFO    InitState.cpp:491            ESP32 chip rev.: 0
68         INFO    InitState.cpp:492            ESP32 SDK version: v4.4.6-dirty
99         INFO    InitState.cpp:494            Wifi MAC: 80:65:99:C8:4D:CC
99         INFO    InitState.cpp:495            LwIP version: 2.1.2esp
E (130) i2c: i2c_set_pin(875): scl gpio number error
[   416][E][esp32-hal-i2c.c:88] i2cInit(): i2c_param_config failed
131        FATAL   InitState.cpp:142            Couldn't initialize two-wire.
141        INFO    ErrorState.cpp:75            Going in error state.
152        INFO    ErrorState.cpp:86            Low level error E2.
160        INFO    TopicHandlerService.cpp:84   Topic handler service stopped.
161        INFO    SettingsService.cpp:335      Settings service stopped.
171        INFO    AudioService.cpp:126         Audio service stopped.
171        INFO    MqttService.cpp:132          MQTT service stopped.

assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace: 0x403781da:0x3fcebcc0 0x4037e00d:0x3fcebce0 0x40383f5d:0x3fcebd00 0x4037f00a:0x3fcebe30 0x420091fa:0x3fcebe70 0x4201783a:0x3fcebe90 0x4204f692:0x3fcebee0

  #0  0x403781da:0x3fcebcc0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037e00d:0x3fcebce0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137     
  #2  0x40383f5d:0x3fcebd00 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4037f00a:0x3fcebe30 in xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1549 (discriminator 1)
  #4  0x420091fa:0x3fcebe70 in ButtonDrv::registerObserver(IButtonObserver&) at src/Hal/ButtonDrv.cpp:186
  #5  0x4201783a:0x3fcebe90 in setup() at src/main.cpp:194
  #6  0x4204f692:0x3fcebee0 in loopTask(void*) at C:/Users/Dominique/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42   

ELF file SHA256: f45148824ef13434

Rebooting...
```

BlueAndi, 2024-04-23:
According to the [board description](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) it has an RGB LED on pin 48 as builtin. But RGB LEDs are currently not supported as status LED, therefore I set it to NC (not connected).

The I2C pins are invalid ... I updated the pinning a little bit:
* Onboard LED: NC
* I2C SDA: 8
* I2C SCL: 9
* I2S WS: 10
* I2S SC: 12
* I2S DI: 13
* LED matrix out: 7
* LDR in: 6

Please pull and check again.

BlueAndi, 2024-04-23:
Schematic: https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-DevKitC-1_V1.1_20221130.pdf

d0m1n1qu3, 2024-04-23:
ok this pin setting doesent work at all

```
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
```

d0m1n1qu3, 2024-04-23:
on this image, from the amazon link above.. it could be that the flash or psram is using some pins .. maybe?

BlueAndi, 2024-04-23:
I choose pin 7 explicit to avoid any internal usage for the LED matrix output. Can you show the complete log, not just the one from the NeoPixelBus library?

BlueAndi, 2024-04-23:
[image]

BlueAndi, 2024-04-23:
Can you try pin 47 for LED matrix out?

d0m1n1qu3, 2024-04-23:
i played around with so much pin combinations an got no working one 

is ther a way to disable I2S and I2C and all other things... i actually need only the led matrix .. maybe that could help?

strange things while i writing this here .. i cut the serial connection because i want to copy paste the text from terminal without scolling ... but now the display works .. WTF

with this pins

```
    -D CONFIG_PIN_ONBOARD_LED=IoPin::NC
    -D CONFIG_PIN_BUTTON_OK=4U
    -D CONFIG_PIN_BUTTON_LEFT=IoPin::NC
    -D CONFIG_PIN_BUTTON_RIGHT=IoPin::NC
    -D CONFIG_PIN_DHT_IN=5U
    -D CONFIG_PIN_I2C_SDA=8U
    -D CONFIG_PIN_I2C_SCL=9U
    -D CONFIG_PIN_TEST=IoPin::NC
    -D CONFIG_PIN_I2S_WS=10U
    -D CONFIG_PIN_I2S_SC=12U
    -D CONFIG_PIN_I2S_DI=13U
    -D CONFIG_PIN_LED_MATRIX_OUT=47U
    -D CONFIG_PIN_LDR_IN=6U
    -D CONFIG_PIN_BATTERY_IN=IoPin::NC
    -D CONFIG_PIN_BUZZER_OUT=IoPin::NC
    -D CONFIG_SENSOR_LDR=SensorLdr::LDR_TYPE_GL5528
    -D CONFIG_SENSOR_LDR_SERIES_RESISTANCE=1000.0F
    -D CONFIG_BUTTON_CTRL=1
    -D CONFIG_LED_TOPO=ColumnMajorAlternatingLayout
    -D CONFIG_SUPPLY_CURRENT=3500U
    -D CONFIG_RTC=0
```

and serial connection on, it starts up and crash with this messages

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

18         INFO    InitState.cpp:486            PIXELIX starts up ...
18         INFO    InitState.cpp:487            Target: esp32-s3-devkitc-1-n16r8v
28         INFO    InitState.cpp:488            SW version: v7.2.1:lc:b
48         INFO    InitState.cpp:490            SW revision: 725eacbb107aa945be2797f752bac328fa06efd6
48         INFO    InitState.cpp:491            ESP32 chip rev.: 0
58         INFO    InitState.cpp:492            ESP32 SDK version: v4.4.6-dirty
89         INFO    InitState.cpp:494            Wifi MAC: 80:65:99:C8:4D:CC
89         INFO    InitState.cpp:495            LwIP version: 2.1.2esp
127        DEBUG   ButtonDrv.cpp:310            ButtonDrv task is ready.
[   429][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ha_dp NOT_FOUND
159        INFO    TopicHandlerService.cpp:73   Topic handler service started.
170        INFO    SettingsService.cpp:328      Settings service started.
170        INFO    AudioDrv.cpp:143             Audio driver task is up.
612        INFO    AudioDrv.cpp:189             I2S driver installed.
621        INFO    AudioService.cpp:103         Audio service started.
[   461][E][Preferences.cpp:483] getString(): nvs_get_str len fail: mqtt_broker_url NOT_FOUND
[   466][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
644        INFO    MqttService.cpp:117          MQTT service started.
655        INFO    ClockDrv.cpp:89              No RTC is available.
[   494][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timezone NOT_FOUND
[   494][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ntp_server NOT_FOUND
[   502][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
974        INFO    SensorDataProvider.cpp:418   Sensor GL5528: available
974        INFO    SensorDataProvider.cpp:418   Sensor SHT3x: -
984        INFO    SensorDataProvider.cpp:418   Sensor DHT11: -
[... 446 lines cut ...]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]    
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
     DEBUG   DisplayMgr.cpp:1312          UpdateTask is up.
5053       DEBUG   DisplayMgr.cpp:1439          [ 35, 35, 36 ]
5073       DEBUG   DisplayMgr.cpp:1451          [  0,  0,  0 ] [ 34, 34, 35 ] [ 34, 34, 35 ]
5063       INFO    DiESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]    
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
```

d0m1n1qu3, 2024-04-23:
but why ther a millon of this messages ...

```
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037dfeb
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
```

d0m1n1qu3, 2024-04-23:
and yes .. it slows down  .. if i wait 10 minutes i get the hello world screen .. but i can not interakt with the pixelix because it is busy by printing the messages

BlueAndi, 2024-04-23:
Because the rmt_wait_tx_done() didn't respond in time. The NeoPixelLibrary choose RMT as default for esp32 S3 chips, the others are running with I2S.

d0m1n1qu3, 2024-04-23:
with small the display show nothing anymore and the messages also appears

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

18         INFO    InitState.cpp:486            PIXELIX starts up ...
18         INFO    InitState.cpp:487            Target: esp32-s3-devkitc-1-n16r8v
28         INFO    InitState.cpp:488            SW version: v7.2.1:lc:b
49         INFO    InitState.cpp:490            SW revision: 725eacbb107aa945be2797f752bac328fa06efd6
49         INFO    InitState.cpp:491            ESP32 chip rev.: 0
59         INFO    InitState.cpp:492            ESP32 SDK version: v4.4.6-dirty
90         INFO    InitState.cpp:494            Wifi MAC: 80:65:99:C8:4D:CC
90         INFO    InitState.cpp:495            LwIP version: 2.1.2esp
128        DEBUG   ButtonDrv.cpp:310            ButtonDrv task is ready.
159        INFO    TopicHandlerService.cpp:73   Topic handler service started.
159        INFO    SettingsService.cpp:328      Settings service started.
169        INFO    ClockDrv.cpp:89              No RTC is available.
[   423][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timezone NOT_FOUND
[   428][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ntp_server NOT_FOUND
[   437][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
486        INFO    SensorDataProvider.cpp:418   Sensor GL5528: available
486        INFO    SensorDataProvider.cpp:418   Sensor SHT3x: -
496        INFO    SensorDataProvider.cpp:418   Sensor DHT11: -
496        INFO    SensorDataProvider.cpp:418   Sensor Battery: -
507        INFO    RestApiTopicHandler.cpp:102  Register: /rest/api/v1/sensors/2/illuminance
[   761][E][Preferences.cpp:483] getString(): nvs_get_str len fail: hostname NOT_FOUND
524        DEBUG   BrightnessCtrl.cpp:294       Light: 0.942 (b-thr 1.036 < x < d-thr 0.753)
524        DEBUG   BrightnessCtrl.cpp:313       Change brightness goal to 241.
534        DEBUG   DisplayMgr.cpp:1250          ProcessTask is up.
534ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037de37
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037de37
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
expression: rmt_wait_tx_done(_channel.RmtChannelNumber, 0)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037de37
file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/NeoPixelBus/src/internal/methods/NeoEsp32RmtMethod.h" line 599
func: bool NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::IsReadyToUpdate() const [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]
```

d0m1n1qu3, 2024-04-23:
> Because the rmt_wait_tx_done() didn't respond in time. The NeoPixelLibrary choose RMT as default for esp32 S3 chips, the others are running with I2S.

and could we fix that .. may be this is the main problem?

BlueAndi, 2024-04-23:
> > Because the rmt_wait_tx_done() didn't respond in time. The NeoPixelLibrary choose RMT as default for esp32 S3 chips, the others are running with I2S.
> 
> and could we fix that .. may be this is the main problem?

I need to reproduce it on my side, then I can contact the maintainer of the NeoPixelBus library as well. Unfortunately I don't have this board here, but I will try with a similar esp32 S3 one.

d0m1n1qu3, 2024-04-23:
we can also have a coding session via discord for example .. but than better in german language ;-)

closed 2024-04-25 by BlueAndi