BlueAndi/Pixelix #172: [Feature] Make analogue clock customizable

nhjschulz, 2024-08-08:
64x64 pixel support feature an analogue clock for the DateAndTime plugin.
But the way it is drawn is hardocded. You cannot change colors or the seconds indication.

Drawing properties  like colors, and mode shall be accessible over REST / WEB UI.  
UI shall make it clear that this is only foe 64x64 displays.

nhjschulz, 2024-08-11:
Can we add a (REST?) call to query the resolution from JS in the plugin-HTML pages ? THe page can then present view dependen options. From a concept perspective I imagine 2 Settings groups, a "generic" and a resolution dependent on below it(if needed)

Someting like this

```
function getViewResolution(pluginUid) {
                disableUI();
                return utils.makeRequest({
                    method: "GET",
                    url: "/rest/api/v1/display/getinfo,
                    isJsonResponse: true
                }).then(function(rsp) {
                   configureUI(rsp)
                }

function configureUI(rsp) {
             if (rsp.resolution.x >= 64 && rsp.resolution >= 64){
                    setVisible( options64x64);
                    enable(options64x64)
              }
}
```

BlueAndi, 2024-08-11:
> Can we add a (REST?) call to query the resolution from JS in the plugin-HTML pages ?

Yes, this can easily be done.
Consider that the plugin itself has to differeniate as well. In the OpenWeatherPlugin I used a static function with a constexpr return parameter to decide at compile whether the forecast feature is enabled or not. This is very important as we are with the 4 MB flash modules over 99%.

closed 2024-12-04 by BlueAndi