BlueAndi/Pixelix #178: [Bug] DateTimePlugin lamp color configuration always the default

ryannevell, 2024-09-16:
**Describe the bug**
It does not appear to be possible to change the color of the lamps used in the DateTimePlugin.

**To Reproduce**
Steps to reproduce the behavior:
1. Install DateTimePlugin into a slot
2. Plugins→DateTimePlugin
3. Change value of "Color of actual day" or "Color of other days"
4. Click "Update"
5. No color change observed

**Expected behavior**
The lamp colors should reflect the colors entered in the settings page.

**Please complete the following information:**
 - Development board: ESP32-DevKit-v1
 - Version: v7.2.4

ryannevell, 2024-09-16:
I think this may be happening because the colors are assigned to the lamp objects in `DateTimePlugin::start`, and remain cached there. So even after `DateTimePlugin::setConfiguration` changes `m_dayOnColor` and `m_dayOffColor`, those new values are not propagated to the 7 `LampWidget` instances.

BlueAndi, 2024-09-16:
Can you please check with the Development branch, it contains the latest code base which will be prepared for v8.0.0?

ryannevell, 2024-09-17:
I cannot reproduce this on latest Development branch. Closing

closed 2024-09-17 by ryannevell