BlueAndi/Pixelix #185: [Feature] Support for ESP-C3

amadea-system, 2024-09-24:
Is there any possibility we could get support for the ESP-C3? I have one these boards: https://magwled.com/products/dgrz-led-cpd-36-1-mini and I would absolutely love to be able to run this project on it.

Thanks!

BlueAndi, 2024-09-24:
Is there a schematic or a board specific pinout somewhere? Couldn't find it on their homepage.

The only specific pins I found:
* GPIO9 - Button
* GPIO20 - Serial RX
* GPIO21 - Serial TX
* GPIO2 - LED data

On the U13 block there is:
* GPIO6
* GPIO7
* GPIO10

amadea-system, 2024-09-24:
Unfortunately I wasn't able to find a full schematic or pinout either. The only other pin that I'm aware of is GPIO 5, which can cut power to the LED strip.

I did spend some time this weekend trying to get it to work myself, But I am far from knowledgeable w/ ESP32 development. There were a few minor changes that I had to make in the code (Add `CONFIG_IDF_TARGET_ESP32C3` to the defined checks in `ResetMon::getResetReasonToStr()`, A type issue in ` AsyncClient::getNoDelay()`, & another typing issue in `AsyncWebSocketClient::remoteIP()`). Afterwards I was able to get the firmware to compile, but the board would only boot loop.

``` text
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf91a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6ac
load:0x403ce710,len:0x25f4
entry 0x403cc710
 ```
 
 I figured I'm missing something a lot more fundamental.

BlueAndi, 2024-09-24:
You can try this branch: https://github.com/BlueAndi/esp-rgb-led-matrix/tree/support/esp32c3

Choose the magwled-1-LED-32x8 environment and try out. Can't promise that it will work!
Additional you need to change again 3rd party libs, to get it compiled.

[image]

[image]

BlueAndi, 2024-10-04:
@amadea-system Any success?

BlueAndi, 2024-10-13:
@amadea-system Is it still an issue or can I close it?

BlueAndi, 2024-10-19:
Closing becaues of no response.

closed 2024-10-19 by BlueAndi