BlueAndi/Pixelix #203: [Feature] Turn on/off display via a timer

Leeefy, 2024-12-18:
basically have an option to turn on/off the display at certain times (eg. 10am mon-fri turn on, 6pm mon-fri turn off).

BlueAndi, 2024-12-18:
Interesting idea to turn it on/off by a timer. Because the development boards with 4 MB flash are full, next year we have to think about how to get some space back to be able to implement more general features.

BlueAndi, 2024-12-27:
Example:
[image]

BlueAndi, 2024-12-29:
The development branch contains now the TimerService, which can be enabled in the config's. For the 8/16 MB flash modules its enabled by default. Note, that the development branch requires already to update the filesystem too.

Please verify.

Leeefy, 2024-12-30:
I have a 4MB flash and I do not see an option for that anywhere, is it not implemented for the 4MB flash modules?

BlueAndi, 2024-12-30:
> I have a 4MB flash and I do not see an option for that anywhere, is it not implemented for the 4MB flash modules?

You can add it, by removing the ";" in front of the used configXXXX.ini file.

See here the part out of the config/configSmall.ini
```
    ...
    # ********** Services (keep the order) **********
    SettingsService @ ~0.1.0 # Mandatory, can not be removed.
    ;MqttService @ ~0.1.0
    TopicHandlerService @ ~0.1.0 # Mandatory, can not be removed.
    FileMgrService @ ~0.1.0 # Mandatory, can not be removed.
    ;AudioService @ ~0.1.0
    ;TimerService @ ~0.1.0 <---------------- Remove ";" in front here.
    ...
```

Remove one ore more plugins to get some space back.

Leeefy, 2024-12-30:
Nevermind, I found it. It doesn't seem to work reliably on my end, and the settings for the service doesnt save between reboot. I also want to suggest to have an option to turn on and off the display from somewhere on the website (perhaps under display tab?) (doesnt make sense to have an option to turn on/off the display via a timer without having an option to turn on/off the screen directly)

BlueAndi, 2024-12-30:
> Nevermind, I found it. It doesn't seem to work reliably on my end, and the settings for the service doesnt save between reboot.

Hmm ... will check it again later.

> I also want to suggest to have an option to turn on and off the display from somewhere on the website (perhaps under display tab?) (doesnt make sense to have an option to turn on/off the display via a timer without having an option to turn on/off the screen directly)

The display can be switched on and off via physical buttons, I thought thats enough. :-)
https://github.com/BlueAndi/Pixelix/tree/Development?tab=readme-ov-file#user-interface

But yes, a button can be added.

BlueAndi, 2024-12-30:
BTW can you load the config/timerService.json file to the file editor in the development web page once after you edit it. And then after reboot. Are they equal or are your configurations lost?

Leeefy, 2024-12-30:
> The display can be switched on and off via physical buttons, I thought thats enough. :-)

Thing is, I have no physical button on my clock lmao (I just have 2 bare wires inside the clock just incase)

> BTW can you load the config/timerService.json file to the file editor in the development web page once after you edit it. And then after reboot. Are they equal or are your configurations lost?

before reboot:
```
...
  "timerSettings": [
    {
      "enabled": true,
      "hour": 1,
      "minute": 20,
...
```

after reboot:
```
...
  "timerSettings": [
    {
      "enabled": false,
      "hour": 0,
      "minute": 0,
...
```

They arent the same.

Leeefy, 2024-12-30:
ALSO while i mention physical button control, (1 button control) holding down the button increases the brightness then decreases it. It doesn't increase a second time, therefore not making it into a loop

BlueAndi, 2024-12-30:
Problem regarding cleared timer settings after reboot should be solved. Please pull from Development branch again.

BlueAndi, 2024-12-30:
> ALSO while i mention physical button control, (1 button control) holding down the button increases the brightness then decreases it. It doesn't increase a second time, therefore not making it into a loop

Can you please open a bug issue for that, so its tracked separately?

Leeefy, 2024-12-30:
> Problem regarding cleared timer settings after reboot should be solved. Please pull from Development branch again.

Seems to save upon a restart now. 

> Can you please open a bug issue for that, so its tracked separately?

Ive opened a new issue for that just now

BlueAndi, 2024-12-30:
Regarding the button in the webpage, I am thinking about providing the possibility not only via REST API, but via MQTT too. It will take some time, because I would like to have a general architectural approach to provide it. The lower interface is clear by using the TopicHandler, which will automatically provide it on REST and MQTT, but in the higher application level its not that clear right now.

BlueAndi, 2024-12-31:
The display can now be powered on/off via REST API and MQTT. A button to the display page was added.

BlueAndi, 2024-12-31:
Update of the REST API on swagger is not ready yet.

BlueAndi, 2025-01-01:
REST API updated: https://app.swaggerhub.com/apis/BlueAndi/Pixelix/1.6.0

BlueAndi, 2025-01-06:
@Leeefy Did you verify successful?

Leeefy, 2025-01-11:
Ive tested everything again (besides api)
Timerservice still doesnt work
The button on the webpage is there but it doesnt work (gets disabled (grays out) but does nothing regarding the clock)
Upon entering the display section, the "Power Off", "Enable auto. display update", "Previous slot", "Next slot", "Next fade effect", "Install" and "Enable" buttons and the " Min. brightness soft limit:" and " Max. brightness soft limit:" are disabled (grayed out) until you interact with something on that webpage (like "Current Brightness:" slider, enabling a plugin slot, etc), then those buttons and sliders becomes enabled
In Development>Debug>Restart, the Restart button doesnt work (gets disabled but doesnt restart)

BlueAndi, 2025-01-12:
Thats sounds like the filesystem was not updated or the browser cache contained the old version.

Leeefy, 2025-01-12:
i will clear browser cache and try again once im home then

Leeefy, 2025-01-13:
cleared cache and it works.
regarding the timeservice service, it did work all this time, i just failed to realise it uses utc+1 instead of utc+0 defined in the datetime plugin

BlueAndi, 2025-01-13:
> cleared cache and it works.

If you update the filesystem, I recommend to always use Shift + F5 in the browser the first time to reload and get an updated cache.

> regarding the timeservice service, it did work all this time, i just failed to realise it uses utc+1 instead of utc+0 defined in the datetime plugin

The timer service uses the specified timezone in the settings. The DateTimePlugin uses this as well if its timezone configuration is empty, otherwise it will only use the plugin specific configured timezone.

> it seems like the brightness% in the timerservice is still broken, 99% is equivelant to 15.3%, 89% = 13.7%, 79% = 12.2%, etc. up until the min cap of 9.8%

I will check it.

BlueAndi, 2025-01-13:
Problem with brightness is fixed, I tested it successfuly. Please verify, only the timer service webpage changed. So you can do a filesystem update or just use the file browser to change it manually.

Leeefy, 2025-01-13:
Tested it, works now!

closed 2025-01-13 by BlueAndi