BlueAndi/Pixelix #209: [Feature] Set fade effect in settings

Leeefy, 2024-12-29:
Basically set the fade effect in settings, as there is currently no way (that im aware about) to have anything but linear fade effect saved upon reboot.
Also potentially have the option to have no fade effect, and have the option to change the fade effect within plugins (like openweatherplugin between temprature, windspeed, humidity and uv-index)

BlueAndi, 2024-12-29:
The default fade effect which is used during changing the active slot can now be changed in the settings. Please verify in in the Development branch.

The text fade effect you see in the OpenWeatherPlugin is a feature of the TextWidget and has no relation to the slot fade effect. The TextWidget only supports the linear effect and is sometimes disabled depended on the plugin.

Leeefy, 2024-12-30:
The new setting seems to work without any issues. Im not sure if its possible but would it be possible to have an option in those plugins to enable/disable the linear effect?

BlueAndi, 2024-12-30:
> The new setting seems to work without any issues.

Ok.

> Im not sure if its possible but would it be possible to have an option in those plugins to enable/disable the linear effect?

In principle possible by adding an additional configuration parameter to every related plugin.

The DateTimePlugin disables the effect to avoid continously fading in case that it shows seconds.
https://github.com/BlueAndi/Pixelix/blob/0b100e94c63fc5ecc52002d4a36d08e4b503721a/lib/Views/src/layouts/DateTimeView32x8.h#L86-L89

Is it so disturbing?

Leeefy, 2024-12-30:
Apologies I havent seen that, sorry about that

BlueAndi, 2024-12-30:
> Apologies I havent seen that, sorry about that

No problem.

Leeefy, 2024-12-30:
Apologies to add to this, but I just tried no fade effect (I forgot to try that yesterday) and when restarting, the pixelix logo stays on screen until power off/restart

BlueAndi, 2024-12-30:
Fixed in Development branch, please update filesystem too.

closed 2024-12-30 by BlueAndi