BlueAndi/Pixelix #210: [Bug] Brightness control with 1 button setup doesnt work as intended

Leeefy, 2024-12-30:
**Expected behavior:**
Holding down the button increases the brightness then decreases it. It should increase a second time, therefore making it into a loop (as described here: https://github.com/BlueAndi/Pixelix/tree/Development?tab=readme-ov-file#user-interface)
**Actual behavior:**
Holding down the button increases the brightness then decreases it. It doesn't increase a second time, therefore not making it into a loop. 

**More Information**
 - Development board: ESP32-DevKit-v1
 - Versions: 8.0.0, 7.2.5, 7.2.4 (Release versions I have used before)

BlueAndi, 2024-12-30:
Fixed in Development branch, please verify.

Leeefy, 2024-12-30:
Works as intended now!

closed 2024-12-30 by Leeefy