BlueAndi/Pixelix #216: API coindesk is offline

ddijanic, 2025-02-16:
the 
http://api.coindesk.com/v1/bpi/currentprice/USD.json
is not workin for 2-3 weeks now
https://api.coinbase.com/v2/prices/BTC-USD/buy
I found a new API, but I can't seem to get it to work. whit GrabViaRestPlugin

BlueAndi, 2025-02-16:
Data from https://api.coinbase.com/v2/prices/BTC-USD/buy looks like e.g.

```json
{"data":{"amount":"97161.665","base":"BTC","currency":"USD"}}
```

to retrieve the amount, the filter should look like

```json
{"data": { "amount": true }}
```

because the amount is a string, you can go with

```"%s $/BTC"```

ddijanic, 2025-02-17:
thanks, cool works,
on old api if I set Format: %0.0F   it shows 96024
now it show 96024.365

I tried that now, but it won't work.
there is a way to shorten a number without a comma

BlueAndi, 2025-02-17:
You can only add the precision to the conversion specifier, like "%.5s $/BTC", which will limit the number of string characters.
It may cause a problem in case the bitcoin rate will be greater or equal than 100000, because you would see only the first 5 characters.

But I guess for now its ok.

BlueAndi, 2025-02-17:
@flcu Did you update the BTCQuotePlugin already, because the current coindesk API seems to be offline?

ddijanic, 2025-02-19:
it works now properly. 
%.5s - work top, and price when gos to 100000 wil be good. 
9x 000 and 100 000. Space or '  it would be good. but he understands that he cannot do that in this plugin

closed 2025-02-20 by BlueAndi